PX4/PX4-Autopilot #12487: Simulating a wifi drone through jMavsim

wuxianshen, 2019-07-16:
I try to simulate a wifi drone following the instructions @
http://dev.px4.io/master/en/simulation/jmavsim.html#simulating-a-wifi-drone
Run "sudo make broadcast jmavsim" at the Firmware directory, but prompting "broadcast cannot be the first argument".
Is there any mistakes in my action or has this feature been removed?

julianoes, 2019-07-18:
@wuxianshen Thanks for raising this issue. This is indeed not working according to the docs anymore.
What exactly are you trying to simulate? Or what is your setup?

@dagar do you know the new make target for this? Or has this been removed?

wuxianshen, 2019-07-18:
@julianoes  Thanks!  I am trying to simulate px4 on one Linux PC and run offboard control code on another windows PC... 
Actually this special requirement does not make real sense ...
Thank you very much !!!

julianoes, 2019-07-18:
@wuxianshen You can also just run the simulation as normal and then set the param `MAV_BROADCAST` to 1 to enable broadcast.

wuxianshen, 2019-07-19:
Thanks @julianoes !
Yeah this really works, but I should use the 14550 port for connecting the uav (No data @ 14540 port), which I thought to be customized for QGC.
So there is nothing different between connection through QGC and offboard control?  We can just use the 14550 port as an alternative for offboard contorl?
Thank you!

julianoes, 2019-07-19:
So you're saying that you would like to have 14540 broadcast as well?

In that case, remove the `-m onboard` from here:
https://github.com/PX4/Firmware/blob/aaad71faab6bdb507bb9525f1c9d64edc33632bc/ROMFS/px4fmu_common/init.d-posix/rcS#L249

Alternatively, to connect to a certain IP you can also add `-t IP` to `mavlink start`.

closed 2019-10-17 by julianoes