PX4/PX4-Autopilot #12490: Rover SITL test failing

ItsTimmy, 2019-07-16:
**Describe the bug**
Some PRs are failing the Rover SITL test, without having changed any functionality related to the rover. For example, [here](http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-12389/7/pipeline/). This is caused by an issue with the timing of the `actuator_controls_0` topic and Gazebo.

**Expected behavior**
The test should pass.

**Log Files and Screenshots**
http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-12389/7/pipeline/

**Drone (please complete the following information):**
- SITL `gazebo_rover`

**Additional context**
This rover test was added recently, in #12117

julianoes, 2019-07-18:
Strange, do you understand what the timing issue is?

ItsTimmy, 2019-07-18:
It seems that, because the system is running in lockstep, Gazebo expects an `actuator_controls_0` message to be published after every sensor reading. In the fixed wing and multicopter attitude controllers, this is no problem, because their timing is driven by polling sensor data. However, `rover_pos_control`'s timing is independent of everything. It's still 250Hz, but perhaps not _exactly_ 250Hz. So there may come a time by random chance where Gazebo wants to provide a second sensor reading before a new `actuator_controls_0` message has come in, but `rover_pos_control` wants to wait an extra millisecond or so to publish the message, so there's a deadlock.

I'll fix this by making the timing depend on polling sensor data.

PR #12519 (2019-07-25): Changed timing of actuator_controls_0 message from rover_pos_control
**Describe problem solved by the proposed pull request**
The `rover_pos_control` module was publishing `actuator_controls_0` messages at a constant 250Hz, independent of any polling on any other UOrb topic. This was causing issues with Gazebo where it would get slightly out of sync with sensor readings. Closes #12490 

**Test data / coverage**
Ran several missions in Gazebo.

**Describe your preferred solution**
Changed `rover_pos_control` to poll on `sensor_combined` to time the publication of `actuator_controls_0`. This sen [... cut]
closed 2019-07-25 by dagar