PX4/PX4-Autopilot #12492: FlightTasks Refactor

Stifael, 2019-07-16:
Given that FlightTasks library does not contain too many tasks yet, it makes sense to do a small refactor rather sooner than late based on the feedback that was received.

Originally the FlightTasks library was designed to simplify the process of adding a new flight maneuver without having to change the already existing tasks. Unfortunately, two implementation strategies evolved:

1.) functionality was shared via inheritance
Initially, it seemed to be a good idea to exploit inheritance for tasks which are similar in nature. However, often only one small part of an existing task needed to change for a "new" task that could not solely be resolved with inheritance but rather was resolved by duplicating the code or add another if-statement. In addition, inheritance made code readability much more difficult.

2.) sharing functionality via utility-classes/functions
This is the preferred way to share functionalities because it simplifies the process of creating new tasks by not having to mess around with existing tasks. 

The current state of the FlightTasks library has the following issues:
- too complicated to trigger a new task (Firmware issue) (1. overwrite existing task vs creating a new PX4-flight mode which requires non-simple changes)
- individual tasks poorly tested 
- duplicated code (FlightTaskAuto + its children)
- some parameters are consumed by different task but require different tuning
- tasks for the same purposes have a different implementation in terms of behavior (FlightTaskAutoLine passes waypoints with full speed if they are located on a straight line vs FlightTaskAutoSmoothVel which stops at every waypoint)
- failsafe is not functional
- several methods that can be overwritten via inheritance makes readability difficult
- tasks cannot stop itself  (was requested during PX4-devsummit)

## short-term suggestions
- keep inheritance to a minimum by only allow the `update`-method to be overwritten
- merge same functionalities of FlightTaskAutoLine / FlightTaskAutoSmoothVel 
- each task that belongs to a specific PX4 flight-mode should behave similarly
- move common functionality into Utility-library
- enable setpoint passing between tasks

MaEtUgR, 2019-07-16:
Sounds good. Some smaller changes into that direction are already in https://github.com/PX4/Firmware/pull/12072 and will hopefully not create too much conflicts with someone else starting to refactor. I'll try to really bring it into a mergable state now before it explodes.

hamishwillee, 2019-07-17:
@MaEtUgR I presume the new docs come after this?

Stifael, 2019-07-17:
> Perhaps we should just create a placeholder doc and link your presentation from devday for now?

yep. the refactor will also not happen within one day and will also not make the presentation obsolete. It is more about aligning on an overall structure such that new tasks follow the new idea instead of copying the existing framework.

MaEtUgR, 2020-02-19:
> - merge same functionalities of FlightTaskAutoLine / FlightTaskAutoSmoothVel
is resolved in https://github.com/PX4/Firmware/commit/7e78eed58642833d27795a13fad39547c382bf35

bresch, 2022-08-15:
We fixed quite a couple of issues in the last years, let's re-open an issue with up-to-date tasks

closed 2022-08-15 by bresch