PX4/PX4-Autopilot #12493: How to connect to px4 from offboard control through ESP8266?

wuxianshen, 2019-07-16:
I'd like to establish a connection with px4 and my offboard control program ( based on pymavlink on Linux) through ESP8266, but failed to found vehicle.
ESP8266 is flashed with firmware v1.2.2, and on PC we can connect px4 through QGC by connecting to wifi PixRacer. 
But When I run the test program which has been all ok with simulator, try to connect to 'udp:192.168.4.1:14540', the program failed to found any system.
We tried 192.168.4.1~4.4 and different ports, none worked.
So how can we connect to px4 from offboard control through ESP8266?
Thanks!!!

hamishwillee, 2019-07-17:
This is an ESP8266 issue - not a Firmware issue. It is almost certainly to do with baud rate mismatch between the ESP8266 and the drone. You need to make sure they are the same 

There is a big discussion on this in Slack at the moment: https://px4.slack.com/archives/C52DXKYLV/p1563185873162600?thread_ts=1562814645.114000&cid=C52DXKYLV

@julianoes @JonasVautherin When that all gets resolved, can you help me understand how I need to update the [esp8266 docs](http://docs.px4.io/master/en/telemetry/esp8266_wifi_module.html)?

I think that two changes are needed. Firstly we need to be VERY clear that the baud rate setting for port and firmware must match. Probably a troubleshooting section, linked from the top. 
Also currently we have [this section](http://docs.px4.io/master/en/telemetry/esp8266_wifi_module.html#flashing-the-esp8266-firmware) on flashing the firmware. I think we need to add a link to the ESPxxx docs on setting the rate via the web interface too: https://github.com/dogmaphobic/mavesp8266/blob/master/HTTP.md

Does that make sense?

hamishwillee, 2019-07-17:
@dogmaphobic ESP8266 seems to give us more trouble than any other comms in terms of setup. Is there any way we can make the setup more robust? Auto discovery of the baud rate?

wuxianshen, 2019-07-17:
@hamishwillee  Thanks for your reply but I really don't know how to view the Slack cause I don't have any valid email account...
And I checked the baudrate set of Telem2 which is used for ESP8266, and the baudrate is set to 921600 following instructions.
[image]
[image]
[image]

wuxianshen, 2019-07-17:
Or is there any mistakes for my connection coding?
[image]
For the url I tried udp on 192.168.4.1~4.4 and port for 14540, 14550, 14555...
Thanks you !!!

hamishwillee, 2019-07-17:
@wuxianshen I have no idea about the setup other than what is in the docs, which is why I connected the other guys. 14540 is the correct port.

irsdkv, 2019-07-24:
I'm also have this issue.
nmap result:

```
sudo nmap -sU -p 14555,14540,14550,4560,14545,4565,14557,14540 192.168.4.1

Starting Nmap 7.60 ( https://nmap.org ) at 2019-07-24 18:44 MSK
WARNING: Duplicate port number(s) specified.  Are you alert enough to be using Nmap?  Have some coffee or Jolt(tm).
Nmap scan report for _gateway (192.168.4.1)
Host is up (0.0032s latency).

PORT      STATE         SERVICE
4560/udp  closed        unknown
4565/udp  closed        unknown
14540/udp closed        unknown
14545/udp closed        unknown
14550/udp closed        unknown
14555/udp open|filtered unknown
14557/udp closed        unknown
MAC Address: A2:20:A6:18:06:4F (Unknown)

Nmap done: 1 IP address (1 host up) scanned in 1.86 seconds
```

Params:
```
nsh> param show SER_TEL2_BAUD
Symbols: x = used, + = saved, * = unsaved
x   SER_TEL2_BAUD [787,1031] : 921600
```

JonasVautherin, 2019-07-24:
It seems like everybody I've heard talking about ESP8266 (@wuxianshen, @irsdkv, @unipheas) can connect with QGC, but not with MAVSDK, and apparently not with pymavlink.

* @irsdkv's nmap command shows that ESP8266 is listening on 14555, which I believe is the port used for communication after discovery. 
* @unipheas said that he did not see heartbeats being emitted by ESP8266 from wireshark.

I am not super confident about this, but I believe that a typical "discovery workflow" would go like this:

1. Drone broadcasts on 14550
2. GCS discovers drone and (somehow) gets its port, here 14555
3. GCS starts talking to drone on 14555

That works for QGC or MAVSDK, probably for pymavlink as well.

Now, if the drone does not broadcast (which is suggested by @unipheas test in this case), the GCS could still be told to connect to the drone directly on 14555. I'm quite confident MAVSDK doesn't do that. But maybe QGC does? Could that be possible?

dogmaphobic, 2019-08-11:
>Drone broadcasts on 14550

Yes. It broadcasts _to_ 14550.

>GCS discovers drone and (somehow) gets its port, here 14555

Every UDP packet contains the source address and port. QGC uses the source address and port to respond.

>GCS starts talking to drone on 14555

No, see above.

>But maybe QGC does?

No, it does not. You can manually create links (on any port), but there is nothing magic about 14555.

What I wonder is if everybody is using the same ESP8266 firmware. I haven't updated the "official" one in a very long time:

https://github.com/dogmaphobic/mavesp8266

I see there are now 68 forks off of it. I know the Ardupilot folks forked it and did some work on it to deal with Ardupilot's issues. You would need a consistent firmware in order to identify the source of the issues. Here is Tridge's fork, which he maintains for Ardupilot:

https://github.com/ArduPilot/mavesp8266

unipheas, 2019-08-12:
As for firmware on the ESP8266 I'm using the pre-built binary listed in the PX4 User Guide [here](https://docs.px4.io/master/en/telemetry/esp8266_wifi_module.html#pre-built-binaries)

Here is the issue I created with the MAVSDK https://github.com/mavlink/MAVSDK-Swift/issues/134

That said, I was able to get it to eventually work, but I wasn't sure how it was doing it. I just plugged it in and for some reason it worked. The other day I met with Jinger in China and it would not work. I am trying to use the ESP8266 as a wifi bridge connected to the telemetry radios for longer range.

closed 2019-11-11 by julianoes