PX4/PX4-Autopilot #12496: parrot bebop2 v1.9.1 and mavsdk

david6096, 2019-07-17:
hi,
I installed px4 v1.9.1 release firmware on bebop2,and qgroundcontrol can connected it and worked correctly.but when I connect it with MAVSDK example(for example battery.cpp),it can discovered the drone but heartbeat message is timedout and the program aborted.
some tips about it?

hamishwillee, 2019-07-17:
This area is for bugs in PX4 firmware, not for support questions. Please raise this in https://discuss.px4.io/c/mavsdk

IF this looks like a bug in mavsdk or PX4 then you'll be asked to raise a request in github. 

Cheers

closed 2019-07-17 by hamishwillee

david6096, 2019-07-17:
yes I see,the MAV_PROTO_VER parameter set to 2(Always use version 2),then MAVSDK example connect the drone correctly and heartbeat message can be processed correctly.