PX4/PX4-Autopilot #12499: Bad position control - need help, log included

yoosun116, 2019-07-17:
Hello, i’m very new to using px4 and recently having serious problems in flight test.
I’m using pixhawk 4, px4 firmware v1.9.1 with Qgroundcontrol for with DJI f450 frame.
Additionally, for altitude measurement, i added TFmini Lidar sensor and changed corresponding parameters.
I basically followed the guides in this site and remained all other settings in default.
So i expected my drone to fly well, but as shown in the log below, when i arm and switch to position mode, it actually does not follow my setpoint command at all…
https://logs.px4.io/plot_app?log=fb60b53e-bfec-43f8-890d-4a4b9e94cfe4
I think it needs serious control gain tuning process or something,
Can anyone diagnose this problem?
Any help is appreciated.
Thank you

julianoes, 2019-07-18:
For this sort of issue my guess is usually that your compass heading is not accurate due to magnetic interference or bad calibration.

I suggest that you do a static test (without flying) and make sure the heading of the drone is correct within 10 degrees or so.

julianoes, 2019-10-17:
@yoosun116 did you find anything in a static test?

julianoes, 2020-02-12:
No answer, closing.

closed 2020-02-12 by julianoes