PX4/PX4-Autopilot #12500: preflight check failed, missing sensor during GAZEBO simulation

joeytrib, 2019-07-17:
**Describe the bug**
I am currently using 1.9.1 and trying to do some simulation in GAZEBO. I followed instructions as depicted here:

    https://dev.px4.io/v1.9.0/en/simulation/ros_interface.html

and I can found mavros is successfully connected.

The simulation used to work, but recently I found whenever I tried to take off the drone, I kept getting "FCU: Primary accelerometer not found". I tried reinstalling mavros and rebuilding px4 and I tried other branches such as px4 1.8.0 but I kept getting the same error. Is there a way to fix it?

Thanks for trying to help.

**To Reproduce**
Steps to reproduce the behavior:
1. roslaunch px4 mavros_posix_sitl.launch (mavros is connected by checking mavros/state)
2. commander takeoff
3. above mentioned fault appeared

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**

output from terminal:

`... logging to /home/gishr/.ros/log/fbea1b46-a856-11e9-a518-9cb6d0e498fb/roslaunch-gishr-XPS-15-9560-26534.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gishr-XPS-15-9560:44011/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
[... 226 lines cut ...]
INFO  [logger] logger started (mode=all)
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> [ INFO] [1563357242.454868424, 1.404000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1563357242.455001818, 1.404000000]: IMU: Scaled IMU message used.
[ INFO] [1563357242.455125475, 1.404000000]: IMU: High resolution IMU detected!
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
[ INFO] [1563357243.327946344, 2.276000000]: IMU: Attitude quaternion IMU detected!
[ INFO] [1563357243.416279637, 2.364000000]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1563357243.428354334, 2.376000000]: IMU: Attitude quaternion IMU detected!
[ INFO] [1563357243.435987869, 2.384000000]: IMU: Scaled IMU message used.
[ INFO] [1563357243.436132577, 2.384000000]: IMU: High resolution IMU detected!
[ INFO] [1563357244.423575966, 3.364000000]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1563357244.423665776, 3.364000000]: VER: 1.1: Flight software:     010901ff (f8db8650d0000000)
[ INFO] [1563357244.423721836, 3.364000000]: VER: 1.1: Middleware software: 010901ff (f8db8650d0000000)
[ INFO] [1563357244.423775584, 3.364000000]: VER: 1.1: OS software:         040f00ff (0000000000000000)
[ INFO] [1563357244.423818526, 3.364000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1563357244.423866079, 3.364000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1563357244.423908980, 3.364000000]: VER: 1.1: UID:                 4954414c44494e4f
[ INFO] [1563357249.329036363, 8.272000000]: FCU: Failsafe mode activated
[ INFO] [1563357250.173524668, 9.116000000]: FCU: Failsafe mode deactivated
WARN  [commander] Primary accelerometer not found
[ERROR] [1563357250.390025081, 9.332000000]: FCU: Primary accelerometer not found
`

bkueng, 2019-07-18:
Can you remove the file `/home/gishr/.ros/eeprom/parameters`?

closed 2019-07-18 by joeytrib

AlexWUrobot, 2023-02-06:
I have a similar issue.
My simulation drone can take off, but the IMU y-acceleration is abnormal.
There are a lot of preflight check failed such as IMU, accelerator 
[image]

ref:https://discuss.px4.io/t/simulation-take-off-motion-not-vertical/30705