PX4/PX4-Autopilot #12503: Airspeed_selector follow up issue

RomanBapst, 2019-07-17:
@sfuhrer FYI
This issue is to keep track of some know limitations of the airspeed selector module #12353
Most of them are related to a setup with multiple airspeed sensors which is not yet supported in PX4.

1) Make airspeed sensor detection and airspeed validator allocation dynamic
There is a potential startup race condition when checking for orb_exists(ORB_ID(airspeed), i).

2) Handling of airspeed scale factor across multiple airspeed validator instances. From which airspeed validator do we take the scale...

3)  When airspeed sensor has been declared invalid we need a better evaluation for when it's ok to use it again. One corner case could be e.g. when the wind estimator just did a reset but no  heading change has taken place. In that case the innovations can be small even if the measured airspeed it totally off because the estimator initialized wind states to make everything consistent.

sfuhrer, 2019-10-15:
unrelated to https://github.com/PX4/Firmware/pull/12887, so this is still valid

julianoes, 2021-02-09:
@sfuhrer is this still up-to-date?

sfuhrer, 2021-02-09:
> @sfuhrer is this still up-to-date?

Kind of. But only relevant for non-default cases (multi airspeed, re-enabling a sensor after failure detected), so I would close this now as we anyway won't work on it soon.

closed 2021-02-09 by sfuhrer