PX4/PX4-Autopilot #12504: EKF replay doesnt reproduce in-flight results with default parameters

abishu, 2019-07-17:
I am trying to use the ekf2 replay as in: https://dev.px4.io/v1.8.2/en/debug/system_wide_replay.html
**PX4 verison:** v1.8.2

My **orb_publisher.rules**:
restrict_topics: sensor_combined, vehicle_gps_position, vehicle_land_detected
module: replay
ignore_others: false

**replay_params.txt**: untouched with flight log.

**Log file type**: ulog.

**command run:** make posix none

The replay estimate and real flight values don’t match. On plotting estimator ->state 1 for instance:
[image]
[image]

abishu, 2019-07-17:
I also posted in: https://discuss.px4.io/t/ekf-replay-doesnt-reproduce-in-flight-results-with-default-parameters/11983

Not sure which is mostly used. Sorry about that.

abishu, 2019-07-17:
Also I had a huge timeslip approx 8-10s after starting the replay!!
[image]

julianoes, 2019-07-18:
@CarlOlsson maybe you could help here?

CarlOlsson, 2019-07-18:
I'm on vacation for two weeks can take a look when I am back. Please upload the logfile which you are trying to replay

bresch, 2022-08-15:
Replayed evolved quite a bit now, not worth investigating an old issue. Please re-open a new issue if needed.

closed 2022-08-15 by bresch