PX4/PX4-Autopilot #12508: LEDDAR_ONE not functional with Github source code

tomcattigerkkk, 2019-07-18:
**Describe the bug**
With the official firmware, one can correctly operate the Leddar one distance sensor with the right setting of the parameter SENS_LEDDAR1_CFG. 

But if I build the firmware with the Github resource and burn the firmware to the same device, there is no distance result with the same setting. In NSH I can see the leddar_one function is started, but there is no reading. 

Could anyone provide some suggestion? Thanks

julianoes, 2019-07-18:
@tomcattigerkkk please provide more info about which version you flash and which version you compile yourself and also the command you use to compile the firmware.

mcsauder, 2019-07-18:
@tomcattigerkkk and @julianoes , this issue appears after my recent refactor of this driver.  When connected to ttyS1 or ttyS2, (telem 1 or telem 2), the driver shows read errors and `collect()` timeouts.  Everything is fine on the default port, ttyS3.  I'll work on understanding the breakage and fix it as quickly as I can.

My apologies for the inconvenience.  Thank you for reporting the bug @tomcattigerkkk .

```
Starting LeddarOne Rangefinder on /dev/ttyS1
INFO  [leddar_one] LeddarOne initialized
INFO  [leddar_one] driver started
Starting Main GPS on /dev/ttyS0
Conflicting config for /dev/ttyS1
```

tomcattigerkkk, 2019-07-18:
> @tomcattigerkkk please provide more info about which version you flash and which version you compile yourself and also the command you use to compile the firmware.

Thank you for your suggestion. 
I use the Pixhawk 2 Cube with the master branch. I've tried v1.9.0 and 1.9.2, both do not work. 
I use `make px4_fmu-v3_default (upload)` to compile and upload the firmware. I've also tried to use QGC to update the compiled firmware. Nothing changes.

mcsauder, 2019-07-18:
Hi @tomcattigerkkk , v1.9.0 and v1.9.2 do not contain my recent refactor of the driver, so perhaps I am not encountering your issue exactly.  I do not have an fmu-v3 to test with, but I'll continue working with the driver to see what I can learn.

tomcattigerkkk, 2019-07-19:
@mcsauder 
I have just tested with the old version of leddar_one.cpp. It works. So, it is because of the new version of the leddar_one code.

mcsauder, 2019-07-19:
Thanks for your update @tomcattigerkkk , I'm working on getting it fixed.  I'll update you as I make progress.

mcsauder, 2019-08-30:
@tomcattigerkkk , let me know if you can test PR #12508 and if it fixes this issue.  Everything is working as it should for me with that PR.  I hope it works for you as well and this fixes the breakage I created.  My apologies for the delay.