PX4/PX4-Autopilot #12509: Unable to build px4 native on beaglebone blue

Kirito1136, 2019-07-18:
Intro
Because i need to do some extra calculations in the air i decided to buy a Beaglebone blue to use it as a flight controller in my project.

Bug Description.
I used the guide that is written on docs.px4.io to upload the firmware native at the Beaglebone blue.
https://docs.px4.io/v1.9.0/en/flight_controller/Beaglebone_blue.html
I believe that the documentation is outdated by my time of writing because the command
/Firmware$ make posix_bbblue_native 
i cannot findit so instead i used the command 
/Firmware$ make beaglebone_blue_native

but that asside.
the problem is that at 2-20% its giving the error: 
'MODULE_NAME' undecleared (first use in this function)

To Reproduce this error 
1 connect the beaglebone to a laptop
2 flash BBB-blank-debian-9.5-iot-armhf-2018-10-07-4gb on it
3 install extra packages that are needed > apt-get install numpy, toml, ect
4 clone the librobotcontrol from the website (2.0.4)
5 rename the librobotcontrol to LIBROBOTCONTROL_INSTALL_DIR
6 clone the px4 firmware (1.9.0) to home/Firmware (standard location)
7 Run: debian@beaglebone:\~/Firmware$ sudo make beaglebone_blue_native
8 Between 2 and 20% see the magic happens

Expected behavior
Go on with the installation and finish it so that px4 can be used on the beaglebone

Log files of the installation

debian@beaglebone:\~/Firmware$ sudo make beaglebone_blue_native
[  1%] Built target df_bmp280
[  1%] Built target df_mpu9250
[  1%] Built target uorb_headers
[  3%] Built target df_driver_framework
Scanning dependencies of target drivers_board
Scanning dependencies of target rc
[  3%] Built target mixer_gen_6dof
[  3%] Building C object boards/beaglebone/blue/src/CMakeFiles/drivers_board.dir/init.c.o
[  3%] Built target mixer_gen
[  3%] Building CXX object src/lib/rc/CMakeFiles/rc.dir/crsf.cpp.o
[  3%] Built target ver_gen
[  3%] Built target git_gps_devices
[  3%] Built target git_ecl
[  3%] Built target git_mavlink_v2
[  3%] Built target git_driverframework
[  4%] Built target lockstep_scheduler
Scanning dependencies of target tinybson
[  4%] Building CXX object src/lib/parameters/tinybson/CMakeFiles/tinybson.dir/tinybson.cpp.o
Scanning dependencies of target uorb_msgs
[  4%] Building CXX object src/lib/rc/CMakeFiles/rc.dir/st24.cpp.o
In file included from /home/debian/Firmware/boards/beaglebone/blue/src/init.c:40:0:
/home/debian/Firmware/boards/beaglebone/blue/src/init.c: In function ‘rc_init’:
/home/debian/Firmware/src/platforms/px4_log.h:232:29: error: ‘MODULE_NAME’ undeclared (first use in this function)
   px4_log_modulename(level, MODULE_NAME, fmt, ##__VA_ARGS__); \
                             ^
/home/debian/Firmware/src/platforms/px4_log.h:388:29: note: in expansion of macro ‘__px4_log_modulename’
 #define PX4_INFO(FMT, ...)  __px4_log_modulename(_PX4_LOG_LEVEL_INFO, FMT, ##__VA_ARGS__)
                             ^~~~~~~~~~~~~~~~~~~~
/home/debian/Firmware/boards/beaglebone/blue/src/init.c:56:2: note: in expansion of macro ‘PX4_INFO’
  PX4_INFO("Initializing librobotcontrol ...");
  ^~~~~~~~
compilation terminated due to -Wfatal-errors.
boards/beaglebone/blue/src/CMakeFiles/drivers_board.dir/build.make:62: recipe for target 'boards/beaglebone/blue/src/CMakeFiles/drivers_board.dir/init.c.o' failed
make[3]: *** [boards/beaglebone/blue/src/CMakeFiles/drivers_board.dir/init.c.o] Error 1
CMakeFiles/Makefile2:9857: recipe for target 'boards/beaglebone/blue/src/CMakeFiles/drivers_board.dir/all' failed
make[2]: *** [boards/beaglebone/blue/src/CMakeFiles/drivers_board.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  4%] Building CXX object src/lib/rc/CMakeFiles/rc.dir/sumd.cpp.o
[  4%] Building CXX object msg/CMakeFiles/uorb_msgs.dir/topics_sources/actuator_armed.cpp.o
[  4%] Linking CXX static library libtinybson.a
Makefile:149: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:193: recipe for target 'beaglebone_blue_native' failed
make: *** [beaglebone_blue_native] Error 2

Thanks in advance

bkueng, 2019-07-18:
Hi
Can you replace `PX4_INFO` and `PX4_ERR` in https://github.com/PX4/Firmware/blob/master/boards/beaglebone/blue/src/init.c with a `printf`?

Kirito1136, 2019-07-18:
Hi 
Thank you very much for the quick response.

I did replace the PX4_INFO and PX4_ERR with the printf function.
I also build again the firmware with the command make.
It walked trough the whole installation this time and its still giving errors

Below i put the full instalation log file.

debian@beaglebone:~/Firmware$ sudo make beaglebone_blue_native
[  1%] Built target df_bmp280
[  1%] Built target df_mpu9250
[  1%] Built target uorb_headers
[  3%] Built target df_driver_framework
Scanning dependencies of target drivers_board
[  3%] Built target mixer_gen_6dof
[  3%] Building C object boards/beaglebone/blue/src/CMakeFiles/drivers_board.dir/init.c.o
[  3%] Built target mixer_gen
[  4%] Built target rc
[  4%] Built target ver_gen
[  4%] Built target git_gps_devices
[  4%] Built target git_ecl
[  4%] Built target git_mavlink_v2
[  4%] Built target git_driverframework
[  5%] Built target lockstep_scheduler
[  5%] Built target tinybson
[  5%] Built target perf
[  8%] Built target work_queue
[  8%] Built target ecl_airdata
Scanning dependencies of target ecl_attitude_fw
Scanning dependencies of target ecl_geo
[  8%] Building CXX object src/lib/ecl/geo/CMakeFiles/ecl_geo.dir/geo.cpp.o
[  8%] Building CXX object src/lib/ecl/attitude_fw/CMakeFiles/ecl_attitude_fw.dir/ecl_controller.cpp.o
[ 23%] Built target uorb_msgs
[ 24%] Built target ecl_geo_lookup
[ 25%] Building CXX object src/lib/ecl/attitude_fw/CMakeFiles/ecl_attitude_fw.dir/ecl_pitch_controller.cpp.o
[ 25%] Building CXX object src/lib/ecl/attitude_fw/CMakeFiles/ecl_attitude_fw.dir/ecl_roll_controller.cpp.o
[ 26%] Linking C static library libdrivers_board.a
[ 26%] Building CXX object src/lib/ecl/attitude_fw/CMakeFiles/ecl_attitude_fw.dir/ecl_wheel_controller.cpp.o
[ 26%] Built target drivers_board
Scanning dependencies of target ecl_tecs
[ 26%] Building CXX object src/lib/ecl/tecs/CMakeFiles/ecl_tecs.dir/tecs.cpp.o
[ 26%] Building CXX object src/lib/ecl/attitude_fw/CMakeFiles/ecl_attitude_fw.dir/ecl_yaw_controller.cpp.o
[ 26%] Linking CXX static library libecl_geo.a
[ 26%] Built target ecl_geo
Scanning dependencies of target ecl_validation
Scanning dependencies of target mixer
[ 26%] Building CXX object src/lib/ecl/validation/CMakeFiles/ecl_validation.dir/data_validator.cpp.o
[ 26%] Building CXX object src/lib/mixer/CMakeFiles/mixer.dir/mixer_helicopter.cpp.o
[ 26%] Linking CXX static library libecl_attitude_fw.a
[ 26%] Built target ecl_attitude_fw
[ 26%] Built target version
[ 26%] Building CXX object src/lib/ecl/validation/CMakeFiles/ecl_validation.dir/data_validator_group.cpp.o
[ 26%] Linking CXX static library libecl_tecs.a
[ 26%] Generating listener_generated.cpp
[ 26%] Built target ecl_tecs
[ 27%] Linking CXX static library libmixer.a
Scanning dependencies of target parameters
[ 27%] Built target generate_topic_listener
Scanning dependencies of target ecl_l1
[ 27%] Building CXX object src/lib/parameters/CMakeFiles/parameters.dir/parameters.cpp.o
[ 27%] Built target mixer
[ 27%] Building CXX object src/lib/ecl/l1/CMakeFiles/ecl_l1.dir/ecl_l1_pos_controller.cpp.o
[ 28%] Linking CXX static library libecl_validation.a
[ 28%] Built target ecl_validation
[ 28%] Linking CXX static library libecl_l1.a
[ 28%] Built target ecl_l1
[ 28%] Linking CXX static library libparameters.a
[ 29%] Built target parameters
Scanning dependencies of target px4_layer
[... 716 lines cut ...]
[ 98%] Linking CXX static library libecl_EKF.a
[ 98%] Built target ecl_EKF
Scanning dependencies of target modules__ekf2
[ 98%] Building CXX object src/modules/ekf2/CMakeFiles/modules__ekf2.dir/ekf2_main.cpp.o
[ 98%] Linking CXX static library libmodules__mc_pos_control.a
[ 98%] Built target modules__mc_pos_control
[ 98%] Linking CXX static library libmodules__ekf2.a
[ 98%] Built target modules__ekf2
Scanning dependencies of target px4
[ 98%] Building CXX object platforms/posix/CMakeFiles/px4.dir/src/main.cpp.o
[100%] Building CXX object platforms/posix/CMakeFiles/px4.dir/apps.cpp.o
[100%] Linking CXX executable ../../bin/px4
../../src/lib/DriverFramework/framework/src/libdf_driver_framework.a(I2CDevObj.cpp.o): In function `DriverFramework::                         I2CDevObj::_readReg(unsigned char, unsigned char*, unsigned int)':
/home/debian/Firmware/src/lib/DriverFramework/framework/src/I2CDevObj.cpp:176: undefined reference to `rc_i2c_read_da                         ta'
collect2: error: ld returned 1 exit status
platforms/posix/CMakeFiles/px4.dir/build.make:282: recipe for target 'bin/px4' failed
make[3]: *** [bin/px4] Error 1
CMakeFiles/Makefile2:10195: recipe for target 'platforms/posix/CMakeFiles/px4.dir/all' failed
make[2]: *** [platforms/posix/CMakeFiles/px4.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:193: recipe for target 'beaglebone_blue_native' failed
make: *** [beaglebone_blue_native] Error 2

Thanks in advance

julianoes, 2019-07-18:
> undefined reference to rc_i2c_read_da ta

It seems like this symbol is not available https://github.com/PX4/DriverFramework/blob/06277ef49fb8c9fad18d56ef40e8bc9fe1655a65/framework/src/I2CDevObj.cpp#L176.

but I have no idea where this library would come from.

silentjet, 2019-07-19:
> > undefined reference to rc_i2c_read_da ta
> 
> It seems like this symbol is not available https://github.com/PX4/DriverFramework/blob/06277ef49fb8c9fad18d56ef40e8bc9fe1655a65/framework/src/I2CDevObj.cpp#L176.
> 
> but I have no idea where this library would come from.
> 
> @UAV-Pilot do you know?

It has to be replaced with providing one
[missing_symbol.patch.txt](https://github.com/PX4/Firmware/files/3408853/missing_symbol.patch.txt)

Kirito1136, 2019-07-19:
Thank you very much guys

It was indeed about the line @silentjet pointed out,
Its all up and running

bkueng, 2019-07-19:
@Kirito1136 @silentjet can one of you please submit a PR to fix these? Thanks.

silentjet, 2019-07-23:
PR #239 created

julianoes, 2020-01-24:
I'm going to close this because a lot has changed recently around DriverFramework etc.

closed 2020-01-24 by julianoes

dagar, 2020-01-24:
These issues were fixed and we build `beaglebone_blue_default` in CI now.