PX4/PX4-Autopilot #12514: LIS3MDL Silicon Bug

robertlong13, 2019-07-18:
Sorry, this isn't actually a bug that needs fixing; it's already been worked around in the code. But the comment for the workaround says "This works now, but further investigation to determine why this happens would be good"

https://github.com/PX4/Firmware/blob/aaad71faab6bdb507bb9525f1c9d64edc33632bc/src/drivers/magnetometer/lis3mdl/lis3mdl.cpp#L363-L366

I've confirmed that this is actually a silicon bug (or possible a "feature") in the LIS3MDL, and has nothing to do with the PX4 firmware. In the screenshot attached, you'll see a trace of an I2C read from a LIS3MDL attempting to read 8 registers X_L, X_H, Y_L, Y_H, Z_L, Z_H, T_L, T_H.
[image]

After reading the X, Y, and Z registers, it should continue on to the temperature registers, but instead, it loops back to X. The only way to read the temperature is to read starting from the temperature registers, which is what you've done.

So there you have it---further investigation done. Sorry for opening a bug report for a non-bug. I didn't know how else to open a conversation about this comment in the code. Possibly nobody cares, but figured I'd share my findings.

bkueng, 2019-07-19:
> So there you have it---further investigation done

Great, thanks!

> Sorry for opening a bug report for a non-bug. I didn't know how else to open a conversation about this comment in the code. Possibly nobody cares, but figured I'd share my findings.

This is still valuable information. Can you open a pull request that updates the comment?

themcfadden, 2019-07-22:
I observed the same behavior on the Bosch BMI088. The documentation does not say anything about limits to burst reads, so I was surprised when the temperature wasn't working and I had to add a separate register read. I assumed that since the temperature register is not updated as often, it had to do something with that.

closed 2019-07-22 by mhkabir