PX4/PX4-Autopilot #12515: External INS For Autopilot

will-dillingham, 2019-07-18:
Hello! I am new to the drone game but have doing a lot of research over the last few weeks on integrating an external INS as the main source of IMU and GNSS data for the Pixhawk 2. I have the ability to output data from my INS upon power up, sending data is not my problem. I know I will have to input data via the to SPI bus via DF17 connector, but what I cannot find information on is firmware modifications that need to be made to be able to use (ie re-mapping variables) external INS. My motivation for this is to have a much higher accuracy than the on-board IMU provides for the Pixhawk 2 and to use the GNSS receiver to correct data when it can. Perhaps this is only something that can be done using Ardupilot? Like I said, new to the drone game. Thank you!!

bkueng, 2019-07-19:
Hi!
Which INS is that? Might be interesting for others too, so would be great if you can share the code.
In any case, what you need to do is to publish the sensor data from the driver. Take a look at other existing IMU drivers, you'll need `PX4Gyroscope` to publish gyro data for example.

itsyaboiwillyd, 2019-07-19:
Inertial Labs INS-DL. Here is sensor data from a quick data acquisition test I ran. Let me know if this is not what you meant!

Wouldn't let me upload just a .txt file so here's a zipped folder with just the .txt file.
[INS_Data_Sample.zip](https://github.com/PX4/Firmware/files/3410993/INS_Data_Sample.zip)

itsyaboiwillyd, 2019-07-19:
When you say publish the sensor data from the driver do you mean the pixhawk or the INS? I currently don't actually have the pixhawk, this is something I want to try and figure out before I make the purchase.

bkueng, 2019-07-19:
From the Pixhawk, so the rest of the system uses that sensor data. You will need to write a SPI driver to get the data from the INS. For example you can look at how the MPU6000 driver does it: https://github.com/PX4/Firmware/blob/master/src/drivers/imu/mpu6000/MPU6000.cpp#L889 (that is a rather complex example for an IMU driver though).

itsyaboiwillyd, 2019-08-02:
I'm back! I purchased the Pixhawk 2.1 Cube and installed Mission Planner to see some of the features. Super cool stuff, im learning a lot. Once I create a SPI driver, what other firmware modifications would I need to make? At that point would I be able to use the DF-17 connector to feed data into flight stack? Thanks!

bkueng, 2019-08-05:
> Mission Planner to see some of the features

I suggest using QGroundControl, simply because it's better tested against PX4. Up to you though.

> Once I create a SPI driver, what other firmware modifications would I need to make?

- Once you have the driver you can already start it on the shell and check if it's publishing data.
- make sure it's started on bootup: add to https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v3/init/rc.board_sensors#L16.
- for testing I suggest to remove all other IMU drivers from the startup script (same file as above, all `mpu* start` commands). `sensors status` shows all IMU publishers.

itsyaboiwillyd, 2019-08-06:
Awesome, thank you so much!!

closed 2019-12-05 by mhkabir

jlecoeur, 2019-12-05:
@itsyaboiwillyd have you made progress on this topic?

itsyaboiwillyd, 2019-12-06:
Yes, currently I am outputting INS data packets as NMEA messages. NMEA messages are being send to Pixhawk using the GPS port.

jlecoeur, 2019-12-06:
OK, are you using the external INS for its position estimate only, and not its attitude or angular rate estimates?

itsyaboiwillyd, 2019-12-06:
Position, heading, and velocity using NMEA messages. VTG, HDT, and GGA.

jlecoeur, 2019-12-06:
@itsyaboiwillyd thank you for the info 👍

tpychev, 2024-03-15:
Hello
I have the same problem, INS NMEA via gps port
I connected it as a GPS source ([used this guid ](https://inertiallabs.com/simplified-integration-with-pixhawk/)), but I can’t figure out how to make it work as a course source.
The INS sends a message to the RMC containing compass data. But I haven’t yet found how to use it in atopilot.
When I disable the compass, my rover says there is no compass.

can you tell me or show me your config?

Thank you