PX4/PX4-Autopilot #12516: Could not calibrate magnetometer on BeagleBone Blue

silentjet, 2019-07-18:
**Describe the bug**
Could not calibrate sensors (mag, accel, gyro) on BeagleBone Blue board.
Every calibration ends up with several timeour error messages.
When installing the board on the actual frame of the drone, it is almost impossible to fly due to the huge instability and constant jerking.

**To Reproduce**
Steps to reproduce the behavior:
1. Board powered up
2. px4 service is started manually by using command 
```bash
sudo build/beaglebone_blue_native/bin/px4 -s posix-configs/bbblue/px4.config
```
3. By using WiFi QGC connects to the px4
4. Starting calibration of the magnetometer
5. Calibration failed with a timeouts

**Expected behavior**
Calibration succeed. Drone behaves stable when in the air.

**Log Files and Screenshots**
[calibration_console_log.txt](https://github.com/PX4/Firmware/files/3408674/calibration_console_log.txt)
[image]

**Drone (please complete the following information):**
- standard quad 450

**Additional context**

julianoes, 2019-07-19:
Just FYI, I'm not aware of anyone currently using PX4 on the BeagleBone Blue, so you'll probably have to fix these things yourself.

julianoes, 2019-07-19:
Also, the plan is to deprecate the DriverFramework at some point which would mean some changes for the BBB coming up. On the flipside this would then make thinks like calibration consistent again and prevent these sorts of issues.

closed 2022-08-15 by bresch