PX4/PX4-Autopilot #12522: Support mpxv7002dp airspeed sensor for fixedwings

moreba1, 2019-07-19:
We used the analog airspeed sensor (mpxv7002dp) in arduplane code and we have no problem. The digital version ( 4525DO) costs a lot more and does not make much difference.

Digital 
[image]

Analog
[image]

julianoes, 2019-07-22:
And here is the datasheet: https://www.nxp.com/docs/en/data-sheet/MPXV7002.pdf

Whoever has the need for this sensor or some spare time, go ahead and add support. Pull requests are always welcome.

dagar, 2019-07-29:
Analogue airspeed should already be support on boards like the original pixhawk. You need to set SENS_DPRES_ANSC.

https://docs.px4.io/master/en/advanced_config/parameter_reference.html#SENS_DPRES_ANSC

Although I've personally never used it and I haven't seen anyone use or test it for years, so it could easily be broken.

moreba1, 2019-07-29:
https://discuss.px4.io/t/analog-airspeed-sensor-mpxv7002dp-on-a-pixhawk-issue/6329

QGC and px4 can't detect analog airspeed sensor .

moreba1, 2019-07-30:
I could connect sensor to pixhawk. For calibration we must set offset or scale or both?

aschershapiro, 2019-08-10:
set scale to 1000 for mpxv7002dp, reboot and then do normal airspeed calibration. it will find the offset automatically

hamishwillee, 2025-03-27:
Closing. This is tagged for documentation, but I don't think there is any work to be done here.

closed 2025-03-27 by hamishwillee