PX4/PX4-Autopilot #12523: Get 'LOCAL_POSITION_NED' parameter error through pymavlink ?

wuxianshen, 2019-07-20:
I try to make a offboard control of PX4 with pymavlink, and in the program I need to read 'LOCAL_POSITION_NED' message from PX4, but the results tend to be **fixed** as the first reading, they don't change at all.
Is there any problem with my implementation? Or is it because PX4 support not perfect?
print([self.the_connection.messages['LOCAL_POSITION_NED'].x, self.the_connection.messages['LOCAL_POSITION_NED'].y, self.the_connection.messages['LOCAL_POSITION_NED'].z])

Thank you!!
----------------------------
For why I use pymavlink instead of mavsdk, cause I met a problem of fail to update offboard target point with mavsdk, which is proposed in https://github.com/mavlink/MAVSDK/issues/784...

wuxianshen, 2019-07-20:
I call the request like this, but is this message deprecated?
self.the_connection.mav.request_data_stream_send(self.the_connection.target_system, self.the_connection.target_component, # mavutil.mavlink.MAV_DATA_STREAM_POSITION, 5, 1) mavutil.mavlink.MAV_DATA_STREAM_ALL, 5, 1)

julianoes, 2019-07-22:
If I was you, I'd try to add support for what you need in MAVSDK. I think that's easier than debugging pymavlink.

wuxianshen, 2019-07-23:
@julianoes  Yeah I think so!  MAVSDK seems to be neater and more elegant, but I can fix the problem of prompting "[mc_pos_control] Position-Control Setpoint-Update failed" when use drone indoor ... Pymavlink is not convenient for me to use but can correctly push position and target setpoints.
Then I had to try to use one port (telem1) as the position and target pushing port through pymavlink, another (telem2) as the telemetry subscribing port through MAVSDK.
Maybe there are some problems with the mavsdk passthrough, so pushing position through passthrough and pushing target setpoints through offboard at the same time will cause some communication bug?

julianoes, 2019-07-23:
I suggest that you make pull requests to MAVSDK to fix your issues.

wuxianshen, 2019-07-24:
@julianoes  Sorry I lose a not in last reply, I mean but I can **not** fix the problem of prompting "[mc_pos_control] Position-Control Setpoint-Update failed" when use drone indoor ... 
Haha, very sorry for this mistake...

julianoes, 2019-07-24:
> "[mc_pos_control] Position-Control Setpoint-Update failed"

Hm, why does this happen exactly?

wuxianshen, 2019-07-25:
When use passthrough push vision position and use offboard push target position at the same time, this will happen. MAVSDK is continuously sending message, but I have know idea about the implementation of communication.
By the way, this situation I met in simulation comes with udp connection, I didn't try other real hardware connection types.

julianoes, 2019-07-25:
Ok, if you want someone to look into the MAVSDK issue then I suggest you create an issue there and add all the code that is necessary to reproduce the issue.

closed 2019-11-21 by wuxianshen