PX4/PX4-Autopilot #12524: Enabling custom MAVLINK topic for SITL

yehonathanlitman, 2019-07-20:
I modified Pixhawk's mixer file to subscribe to custom uORB messages, and while it works perfectly fine on a physical quad, it doesn't work in SITL. I can't get the simulator to subscribe to my custom topic, and I'm not sure why. I've tried following the procedure outlined in [here](https://dev.px4.io/v1.9.0/en/middleware/mavlink.html) and applying it directly to
[simulator_mavlink](https://github.com/PX4/Firmware/blob/master/src/modules/simulator/simulator_mavlink.cpp), but I'm not getting seeing the topic in SITL and the quadrotor doesn't subscribe to the messages. I've tried making a custom app to set the multirotor mixer inputs in a different way but while it seems to respond to different parameters, I'm still not able to get the quad to fly properly. If I set the inputs manually the quad can fly perfectly fine, thus it seems like a MAVLINK issue. You can see all my changes [here](https://github.com/yehonathanlitman/Firmware).

If you could refer me in the right direction, I would really appreciate it. Let me know if anything is unclear.

yehonathanlitman, 2019-07-21:
I've found out that `DONT_RUN=1 make px4_sitl gazebo` doesn't include px4io.cpp, therefore my custom values are not stored in the control actuators. Could anyone tell me where the register values for the actuators are getting stored when the simulator is running?

yehonathanlitman, 2019-07-21:
Ok, I figured it out. I didn't understand that SITL uses pwm_out_sim and not px4io, hence why my parameters couldn't go through the `get_control` method. I recommend that this should be specified in the documentation.

closed 2019-07-21 by yehonathanlitman

hamishwillee, 2019-07-21:
>  I recommend that this should be specified in the documentation.

@yehonathanlitman Sounds good. Do you think you could create a PR to update https://dev.px4.io/master/en/middleware/mavlink.html ?