PX4/PX4-Autopilot #12529: Power supply by port 'power 1', then USB port can't used?

keguo1119, 2019-07-22:
I wan't download the log from USB port use mavros, the pixhawk is powered by port 'power 1'. But I find the USB can't used? Does this can be solved ?

[image]

julianoes, 2019-07-22:
This should be possible. What is the error that you're seeing?

Can you try to power by USB first?

julianoes, 2019-07-22:
This could be the same problem as #12201.

mcsauder, 2019-07-22:
@keguo1119 , if you have a chance to test with PR #12428, it might fix your issue.  Please let us know if it does.

keguo1119, 2019-07-26:
@mcsauder ， I have merge the mcsauder:mavlink_config_stream_bugfix_master into current master branch, and test the firmware on pixhawk on fmu-v3. It works fine.   

And now my new question is  do we just need merge the mavlink_main.h mavlink_main.cpp file into the old version 1.8.2, or some other files also need to be change to fix the bug on the old version(like 1.8.2)?

keguo1119, 2019-07-26:
@mcsauder,  could you point out how to fix the bug,  base on version 1.8.2? It's easy to merge the master branch into our current version, but the testing will spend a lot of time.

mcsauder, 2019-07-26:
@keguo1119 , It appears you can backport the same fix to v1.8.2.  I haven't bench tested it, but here is your example: https://github.com/mcsauder/Firmware/tree/v1.8.2_mavlink_config .  If this works for you let us know and we can close out this issue when #12428 gets merged.

keguo1119, 2019-07-26:
@mcsauder , I have modify our own version base on the example, it works fine. Thank you very much.

Otherwise , the key changes is the two functions below, is that right?
[image]

mcsauder, 2019-07-26:
That is correct. Just cut/paste code to create those two methods and allow them to be called from task_main() rather than at the constructor. Glad to hear that worked for you!

keguo1119, 2019-07-29:
@mkschreder 
[image]
It says set CBRK_USB_CHK to 197848  has some risk. what's the potential risk if I disable the  the USB connected checks in the commander. (If not the UAV can't take off)

mcsauder, 2019-08-01:
@keguo1119 , the risk is if the USB link is disconnected during flight it will reboot the autopilot.

@dagar , we can close out this issue once #12428 is merged.

keguo1119, 2019-08-05:
Can we solve this risk in software aspect? I will try to solve this by supply the USB power from the power port!!

mcsauder, 2019-08-05:
Hi @keguo1119 , as luck would have it, #9664 was just merged in, so I think everything should be working to meet your needs.  Can we close out this PR now that #12428 is merged as well?

keguo1119, 2019-08-12:
Yes，this issue can be closed!

closed 2019-11-10 by mcsauder