PX4/PX4-Autopilot #12530: throttle problem

gayass17, 2019-07-22:
hello 
I use pixhawk 2.1 as the hardware and px4 version is 1.9.2.
My fixed-wing plane took off in manual mode. But when I change flight mode into mission, autopilot will shut down throttle. 
What should I do to slove this problem?
This is flight review: https://logs.px4.io/plot_app?log=f6fe68da-652e-46c7-8d1c-758b804a8939
Thanks.

julianoes, 2019-07-22:
@gayass17 is it possible that you don't have an airspeed sensor connected?

[image]

@RomanBapst do you have an idea where these `filtered_airspeed` values come from that we see in the plot?

gayass17, 2019-07-22:
Yeah, another user told me airspeed sensor is not necessary. Should I connect one?

julianoes, 2019-10-25:
FYI @sfuhrer

sfuhrer, 2019-10-26:
@gayass17 did you solve your problem?

Isn't it to be expected that the throttle drops to 0 when your altitude setpoint is well below your vehicle when you switch into mission? If you don't want to have it dropping to 0 you'd have to adapt FW_THR_MIN

julianoes, 2020-02-12:
I'm concerned that we have garbage data like in the plot showing up for airspeed when there is no airspeed sensor connected.