PX4/PX4-Autopilot #12535: Random arming glitches from disarmed

JMshare, 2019-07-22:
**Describe the bug**
It seems like a weird issue, but for a while I had this problem with the logger turning quickly on and off in random intervals when disarmed. Then I noticed it is associated with the arming buzzer noise and I see a quick arming blip in the QGC. The resulting logs can be seen on the screenshot.
It happens only on my big FW airframe with some hungry servos, not on my small quad, so I am thinking it could be some power issue? Or perhaps a SD card problem? I haven't tried swapping them Pixhawks yet which may tell me more.

**To Reproduce**
Basically I just power on the airframe and it starts doing this without me touching anything. 

**Log Files and Screenshots**
[image]

**Drone (please complete the following information):**
I am using the Pixhawk 4 and 1.9.0 PX4 version.
The logging is set to only when armed. 

Any idea what is causing this?

Btw once I arm the system stays armed smoothly without glitches.

I have a log from when I turned logger on manually but it doesn't show anything particular.
[https://review.px4.io/plot_app?log=3c9a3162-22fa-4fa7-bf1e-d3ebe9f95b73](url)

bkueng, 2019-07-22:
This is coming from the RC (you can look at the `input_rc` data). You have mapped an arming switch to channel 16, which shows glitches. The other channels look fine in that log.
Which RC receiver and transmitters are you using?

JMshare, 2019-07-22:
@bkueng well spotted, it's definitely the RC input. 

I am using the Taranis X9D and the RX8R receiver with the XM+ for redundancy. 
I haven't got that problem again since unplugging the XM+.

bkueng, 2019-07-23:
Would be worth looking at the raw rc data then.

julianoes, 2019-07-23:
Closing this as it seems to be an issue of the specific hardware.

closed 2019-07-23 by julianoes