PX4/PX4-Autopilot #12536: Out of Memory Error While Compass Calibration - Here2 CANBus

miracatici, 2019-07-22:
Hello I'm trying to use Here2 GPS module with UAVCAN library. In my first attempt, I flash Pixhawk Cube Black FCU with 1.9.2 stable release binary file via QGC. After flashing, UAVCAN is enabled with Auto-Config mode. After that I did soft reboot (Parameters > Tools > Reboot Vehicle). When reboot is done, GPS status is ok. When I try to calibrate compass sensor, an error is occurred. "Error : out of memory".

Some of the issiues seems to be related. First suggestion is using manual node id configuration instead of auto configuration. However, if I select manual configuration in UAVCAN tab, none of the sensors on the canbus line is appeared. I think this is related with wrong node id. I couldn't find Here2 CANBus Node ID. (Auto configuration says 125 but it doesn't work.)

Is there anyone who faced similar problem? Thanks in advance.

julianoes, 2019-07-23:
@bkueng any advice what to do to avoid running out of RAM?

jzazbert, 2019-12-20:
If this is still relevant... To free up memory, disable any unused mavlink streams, and make sure log from boot is OFF. Use your primary radio to calibrate instead of USB also can save a bunch of RAM. These are all tricks we use to deal with low RAM on cube. Most of these things can be reenabled after calibration.

julianoes, 2020-01-30:
Thanks for the note @jzazbert, I'll close this for now.

closed 2020-01-30 by julianoes