PX4/PX4-Autopilot #12538: Save flight mission in ULOG file and ability to export it

moreba1, 2019-07-22:
Hi

In a professional review of flight for  UAV development or for a crash investigation there is need to know what is total mission and ability to plot it with flight path .

moreba1, 2019-07-22:
https://github.com/PX4/flight_review/issues/165

Antiheavy, 2019-07-22:
agreed.  This would be a great feature to add to the ULOG.  how will you handle multiple mission uploads, for example in the middle of a flight?

moreba1, 2019-07-23:
all of mission must be saved with time of uploading to autopilot

julianoes, 2019-07-23:
@moreba1 pull requests are welcome :)

junwoo091400, 2022-11-22:
This was also discussed in discord server as well: https://discord.com/channels/1022170275984457759/1032055573753118742/1034513943013638144

junwoo091400, 2022-12-01:
Another user requested this again: https://discord.com/channels/1022170275984457759/1022170275984457762/1047648093832757269

julianoes, 2026-08-31:
Would be nice though.