PX4/PX4-Autopilot #12541: EKF2 yaw estimation error

totomyl, 2019-07-23:
Hello,

I am trying to send motion capture data to a pixfalcon running px4 autopilot version 1.8.2. The motion capture system (mocap optitrack) is being published to the topic /mavros/vision_pose/pose. I am currently using EKF2 to estimate the position of the drone, I have set EKF2_AID_MASK to 24 (yaw + vision fusion). The barometer is disabled as well as the magnetometer and gps.

When visualizing the data in RVIZ, I am able to see both the mavros/vision_pose/pose and mavros/local_position/pose topics, however, the local_position topic keeps on yawing while the drone is still. When the drone is being moved around, both the local_position and the vision_pose move but the local_pose is constantly yawing so they are never in line.

I have tried upgrading the px4 firmware to 1.9.2 but this did not solve the problem. The sensors were re-calibrated multiple times and that did not help either.

Any help or guidance would be appreciated, thank you in advance!

RomanBapst, 2019-07-23:
@totomyl You will need to provide a log otherwise we can't help you.

totomyl, 2019-07-23:
@RomanBapst sorry about that, I have added the link to the flight review, I was not sure how to put the actual file onto here as it was not supported.

artifactz, 2019-07-29:
Maybe you can take a look at `rostopich echo /mavros/imu/data_raw` to check if the yawing already happens there. Also, your `EKF2_MAG_NOISE` parameter seems very high (`EKF2_EV_GATE` as well).

julianoes, 2019-07-30:
@totomyl is this a duplicate of  #12577 or something else?

totomyl, 2019-07-30:
> @totomyl is this a duplicate of #12577 or something else?

#12577 is a different problem, the mag won't even disable on it

bresch, 2022-08-15:
This was caused by an issue in the gyro bias estimate (due to some numerical issues in the covariance prediction step). It's now fixed on `main` and release 1.13
[image]

closed 2022-08-15 by bresch