PX4/PX4-Autopilot #12544: Reposition vs Mission Paus

Stifael, 2019-07-23:
To my understanding for both `GoTo` and for` Mission-Pause` the mavlink-msg MAV_CMD_DO_REPOSITION(192) is used. However, there is already a dedicated msg for pausing/continuing a mission called MAV_CMD_DO_PAUSE_CONTINUE (193 ). To be able to distinguish between `GoTo` and `Pause` also within the ulog, it would be beneficial to use 193 for mission-pause instead of 192.

Given that such a change requires changes on both PX4 and QGC, I opened the same issue on QGC as well (https://github.com/mavlink/qgroundcontrol/issues/7612).

julianoes, 2019-07-24:
Notes from dev call:

- @julianoes I would think that change makes sense :smile:

MaEtUgR, 2020-04-02:
This was recently brought up in slack again and there was the idea to use `DO_REPOSITION` with coordinates set to NAN to command a stop. Using the existing pause command would probably be even more desired.

hamishwillee, 2020-04-02:
@MaEtUgR @DonLakeFlyer et al

Using the existing pause message would be **much more desired**. Why don't we do this?

Regarding [MAV_CMD_DO_REPOSITION](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_REPOSITION) hacks, I'm not opposed to this as well as long as the documentation for the message in common.xml gets updated and everyone adopts the same approach. Is this what is expected to happen?

DonLakeFlyer, 2020-04-03:
> Using the existing pause message would be **much more desired**. Why don't we do this?

Do you mean this: MAV_CMD_DO_PAUSE_CONTINUE. First I've seen that. Does it actually work?

hamishwillee, 2020-04-06:
Probably not not on PX4, or this issue would be closed. Looks like we do respond to it in PX4 v1.11: https://github.com/PX4/Firmware/commit/509be8208110476e8d875c94af2c3c6c8c175793#

But implementing this is the "right thing to do" from a mavlink perspective (ie this is the command that meets the intention)

closed 2026-08-15 by github-actions[bot]