PX4/PX4-Autopilot #12546: Boolean parameters not clearly highlighted

hamishwillee, 2019-07-24:
By example, consider [COM_RC_OVERRIDE](http://docs.px4.io/master/en/advanced_config/parameter_reference.html#COM_RC_OVERRIDE) which is defined as a boolean:

```
/**
 * Enable RC stick override of auto modes
 *
 * When an auto mode is active (except a critical battery reaction) moving the RC sticks
 * gives control back to the pilot in manual position mode immediately.
 *
 * Only has an effect on multicopters and VTOLS in multicopter mode.
 *
 * @boolean
 * @group Commander
 */
PARAM_DEFINE_INT32(COM_RC_OVERRIDE, 1);
```

However the rendering doesn't undersand that this is a boolean, and expects a max, min, interval in the left column and units in the right:

[image]

Proposal is to update rendering to make it obvious that a variable is a boolean. Open to ideas. @bkueng What do you recommend? Something like this?
[image]

bkueng, 2019-07-24:
Doesn't QGC show as 'Enabled/Disabled'? If so I'd go for that instead of True/False. And maybe turn it around: `Enabled (1)`.

dagar, 2019-07-25:
After https://github.com/PX4/Firmware/pull/11318 I was planning on introducing a real boolean type at some point.

hamishwillee, 2019-07-25:
@dagar Re "was planning on introducing a real boolean type at some point." - what sort of timeframe? I guess it makes no difference if I do this early.

dagar, 2019-07-26:
Could be sooner if it actually matters for anyone. I viewed it as a small optimization within the param system.

What was your solution going to be @hamishwillee?

Even shorter term as an incremental step we could probably "add" a boolean type that's still using a  full int32 internally.

hamishwillee, 2019-10-24:
> What was your solution going to be 

For me this is just a docs fix - I will do as Beat suggested here https://github.com/PX4/Firmware/issues/12546#issuecomment-514493628. It is posted here because the generation tools are in this repo. 

Whether it is worth implementing it as a boolean is something for you guys (a separate issue).

PR #13269 (2019-10-25): Parameter parser/markdown includes boolean flag
Fixes #12546

Rendering for boolean parameters now shown Enabled/disabled. For example:

[image]
closed 2019-10-25 by LorenzMeier