PX4/PX4-Autopilot #12550: Auto sensor reading after pixhawk startup

bthnekn, 2019-07-24:
Hi guys is there any method for sending messages through mavlink console while pixhawk startup or when pixhawk armed.

ı have a sensor called usharp patch from aerotenna like ulanding radar distance sensor ı am reading data when ı write "usharp_patch start" to the mavlink console and watch the data with analyze section. So ı am wondering can ı send that "usharp_start" message to mavlink console when ı power the pixhawk?

Well, maybe changing flight mode features will do that but ı dont know how to change flight mode features either.

btw sensor's driver is uploaded to the pixhawk but ı dont know who  did that ı'll publish the driver as soon as find the driver file

julianoes, 2019-07-24:
Do I understand correctly that you just want to do a nsh command on startup to start a sensor?

If you're already compiling your own code, you can add the command to the startup here:
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS

bthnekn, 2019-07-25:
Yes just try to add a new command. Normally I write "usharp_patch start" command manually via mavlink console in qground control.What I want is actually this starting process of the sensor triggered with startup like the compass or gyro (sensor dont need calibration btw)

Thank a lot for your help ı will examine your link 

[image]

mhkabir, 2019-07-25:
I'm closing this as a solution has been posted, please use discuss.px4.io if you need further help.

mhkabir, 2019-07-25:
Also see https://dev.px4.io/v1.9.0/en/concept/system_startup.html

closed 2019-07-25 by mhkabir