PX4/PX4-Autopilot #12558: EKF2 not resetting position state when stopping GPS use and EV is already running

artifactz, 2019-07-25:
**Description**
"Failsafe enabled: no local position" when EKF2 switches from GPS to external vision. You can read [my diary on discuss.px4.io](https://discuss.px4.io/t/outdoor-indoor-gps-slam-no-local-position/11958).

**To Reproduce**
Steps to reproduce the behavior:
1. drone with GPS and SLAM/visual odometry/etc., `EKF2_AID_MASK` is GPS + EV_POS + EV_YAW
2. take off, hover/fly for at least 30 seconds (due to some internal EKF2 checks, see forum post)
3. fly into a GPS-denied area (tunnel, hall etc.)

**Expected behavior**
EKF2 recognizes a decrease in GPS quality and switches to EV as primary localization source.
Because GPS was bad and EV can be imperfect as well, it needs to reset its position state, to avoid failing `_vel_pos_test_ratio` checks.

**Log File**
https://logs.px4.io/plot_app?log=b7f5fb21-808a-42eb-844d-18901eb24620

**Screenshot**
[image]

* `pos_test_ratio` explodes as soon as stopping GPS use
* EKF states[7..8] (position) are only reset after dead reckoning and failsafe

**Drone**
- rotary wing

julianoes, 2019-07-26:
@priseborough would be great if you could comment.

mhkabir, 2019-07-26:
@artifactz is your vision yaw in the NED frame?

artifactz, 2019-07-29:
@mhkabir My vision yaw is in ENU, because I use Mavros:

> ROS uses ENU frames as convention, therefore position feedback must be provided in ENU.
(https://dev.px4.io/v1.9.0/en/ros/external_position_estimation.html)

Is that incorrect?

LorenzMeier, 2019-07-31:
@jkflying Could you please assist the review?

closed 2019-09-17 by priseborough