PX4/PX4-Autopilot #12568: EKF ground effect seems not working

hyonlim, 2019-07-28:
**Describe the bug**
See attached log.
topic [vehicle_land_detected_0.in_ground_effect] is 1, however, [estimator_status_0.control_mode_flags], the gnd_effect bit is not 1.
This will make EKF2_GND_EFF_DZ not effective. Basically, innovation deadzone is not working.

**To Reproduce**
Steps to reproduce the behavior:
1. Set EKF2_GND_EFF_DZ to any positive number.
2. It should enter to the clipping function in https://github.com/PX4/ecl/blob/d38164fc8e68b740405669127138e550c3c7375c/EKF/vel_pos_fusion.cpp#L113, however, it won't enter.

**Expected behavior**
Proper operation of ground effect compensation.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:

[log_4_2019-7-28-17-10-08.zip](https://github.com/PX4/Firmware/files/3438964/log_4_2019-7-28-17-10-08.zip)

RomanBapst, 2019-07-28:
@hyonlim Have you seen EKF2_GND_MAX_HGT?
Are you using a lidar?

hyonlim, 2019-07-28:
We are investigating a problem. It looks like #12571 is related. Funny.

hyonlim, 2019-07-28:
Ok. We figured this out.
We made new hardware to test some functionality, and it has i2c device with address 0x42.
PX4Flow app recognize that device regardless of device type.
That makes fake range sensor topic published, and it makes EKF ground effect not usable.

closed 2019-07-30 by julianoes