PX4/PX4-Autopilot #12571: This can lead unnecessary detection of px4flow.

hyonlim, 2019-07-28:
**Describe the bug**
This mandatory execution of px4flow will check i2c bus with address given (0x42), but it does not perform actual sensor identity check. When there is any device with address 0x42, it will recognize it as px4flow and outputs wrong data.

**To Reproduce**
In the hardware put any i2c device with address 0x42

**Expected behavior**
Even though there is a response on the address, it should check whether the device is px4flow or not.

dagar, 2019-07-28:
Let's add a parameter to start px4flow rather than probing every boot.

dagar, 2019-08-05:
PR adding a parameter. https://github.com/PX4/Firmware/pull/12633

hamishwillee, 2019-08-05:
Will need to be added to http://docs.px4.io/master/en/sensor/px4flow.html#pixhawk-setup

closed 2019-08-06 by dagar

Ken-Thorburn, 2019-09-06:
Hey, I can't find the parameter sens_en_px4flow within px4 1.9.2. If this parameter hasn't rolled out yet, I am unable to get the sensor working anyway https://github.com/PX4/Firmware/issues/12911