PX4/PX4-Autopilot #12573: Handle for OPTICAL_FLOW mavlink message

tech0x, 2019-07-28:
**Describe the bug**
I try to connect openmv with optical flow mavlink extension, but px4 firmware does not have OPTICAL_FLOW handler, has only OPTICAL_FLOW_RAD handler. How I can handle OPTICAL_FLOW messages or convert message into OPTICAL_FLOW_RAD format.

Structure have to much differs.
OPTICAL_FLOW_RAD:
uint32_t integration_time_us; /< [us] Integration time. Divide integrated_x and integrated_y by the integration time to obtain average flow. The integration time also indicates the./
float integrated_x; /< [rad] Flow around X axis (Sensor RH rotation about the X axis induces a positive flow. Sensor linear motion along the positive Y axis induces a negative flow.)/
float integrated_y; /< [rad] Flow around Y axis (Sensor RH rotation about the Y axis induces a positive flow. Sensor linear motion along the positive X axis induces a positive flow.)/
float integrated_xgyro; /< [rad] RH rotation around X axis/
float integrated_ygyro; /< [rad] RH rotation around Y axis/
float integrated_zgyro; /< [rad] RH rotation around Z axis/
uint32_t time_delta_distance_us; /< [us] Time since the distance was sampled./
int16_t temperature; /< [cdegC] Temperature/

OPTICAL_FLOW:
float flow_comp_m_x; /< [m] Flow in x-sensor direction, angular-speed compensated/
float flow_comp_m_y; /< [m] Flow in y-sensor direction, angular-speed compensated/
int16_t flow_x; /< [dpix] Flow in x-sensor direction/
int16_t flow_y; /< [dpix] Flow in y-sensor direction/
float flow_rate_x; /< [rad/s] Flow rate about X axis/
float flow_rate_y; /< [rad/s] Flow rate about Y axis/

jlecoeur, 2019-07-29:
Hi @tech0x 

You have two options:
- on OpenMV side, modify the firmware to send OPTICAL_FLOW_RAD messages
- on PX4 side, add a handler for OPTICAL_FLOW messages

For the PX4 side, you need to add a piece of code that converts the mavlink message into a uorb message. You can look at how OPTICAL_FLOW_RAD is handled ([mavlink_receiver.h#L147](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.h#L147), [mavlink_receiver.cpp#L133-L135](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L133-L135), [mavlink_receiver.cpp#L566-L627](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L566-L627)) and do the same for OPTICAL_FLOW. Feel free to create a pull request and ping me. 

Be careful with units: while OPTICAL_FLOW_RAD gives time-integrated values (in radians) and integration timespan (in us), OPTICAL_FLOW gives both raw values (in rad/s and m/s) integrated values (in pixels), but does not indicate the integration timespan. 
@hamishwillee the documented units do not seem correct for OPTICAL_FLOW, `flow_comp_m_x` and `flow_comp_m_y` should be in m/s. Is this a known issue?

hamishwillee, 2019-07-30:
@tech0x With respect to @jlecoeur option 2 "on the PX4 side", there is some further explanation of what is going on/what needs to be done in https://dev.px4.io/master/en/middleware/mavlink.html

> the documented units do not seem correct for OPTICAL_FLOW, flow_comp_m_x and flow_comp_m_y should be in m/s. Is this a known issue?

@jlecoeur Are you sure? The original description was "Flow in meters in x-sensor direction, angular-speed compensated". The only change since that was to move the units out of the description. It seems logical that "m" would be correct because  they are a compensated value of the flow field -which is measured in dezi-pixels (a distance, not a velocity unit). Of course I know nothing at all about flow!

There is certainly no bug report.

jlecoeur, 2019-07-30:
@hamishwillee thanks for pointing to the docs!

I am quite sure about the units. I opened an issue here https://github.com/mavlink/mavlink/issues/1187

hamishwillee, 2019-10-28:
FYI, the optical flow method units got fixed up in https://github.com/mavlink/mavlink/issues/1187

tech0x, 2019-10-28:
Good day, 
Thank Your for all resonses, I will try to make this.

tech0x, 2019-10-28:
> Hi @tech0x
> 
> You have two options:
> 
> * on OpenMV side, modify the firmware to send OPTICAL_FLOW_RAD messages
> * on PX4 side, add a handler for OPTICAL_FLOW messages
> 
> For the PX4 side, you need to add a piece of code that converts the mavlink message into a uorb message. You can look at how OPTICAL_FLOW_RAD is handled ([mavlink_receiver.h#L147](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.h#L147), [mavlink_receiver.cpp#L133-L135](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L133-L135), [mavlink_receiver.cpp#L566-L627](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L566-L627)) and do the same for OPTICAL_FLOW. Feel free to create a pull request and ping me.
> 
> Be careful with units: while OPTICAL_FLOW_RAD gives time-integrated values (in radians) and integration timespan (in us), OPTICAL_FLOW gives both raw values (in rad/s and m/s) and integrated values (in pixels) but does not indicate the integration timespan.
> @hamishwillee the documented units do not seem correct for OPTICAL_FLOW, `flow_comp_m_x` and `flow_comp_m_y` should be in m/s. Is this a known issue?

Hm sounds great, can You recomend something for debuging optflow logic ?

closed 2019-12-05 by mhkabir

jlecoeur, 2019-12-05:
Was this fixed?

mhkabir, 2019-12-05:
@jlecoeur we have the OPTICAL_FLOW_RAD interface as the standard way to interface external flow sensors.

OPTICAL_FLOW is a older (now deprecated for a few years) interface when the velocity computation used to happen on the flow module. It turns out that the navigation EKF is the better place for this to happen since it has all the best data available to it, and yields much higher performance.

You can "make it work" with the old interface, but I do not see much value.

jlecoeur, 2019-12-05:
OK, agreed 👍