PX4/PX4-Autopilot #12577: Cannot disable magnetometer

totomyl, 2019-07-29:
Hi,

I am sending motion capture data to /mavros/vision_pose/pose and want to disable the magnetometer since I am testing indoors. I have set the checked the ekf position fusion and the yaw fusion under the aid mask parameter. then disabled the cal_mag0 and sys_has_mag. but when I do so, my mavros/local_position/pose topic cannot be seen at all.

julianoes, 2019-07-30:
@RomanBapst do you know what is wrong here?

mhkabir, 2019-07-30:
EKF2 doesn't initialize without mag data, this is a known issue. There's nothing fundamentally preventing us from fixing this, it's just a matter of testing the said fix and keeping it maintained.

mhkabir, 2019-07-30:
I can make a PR for this soon.

Antiheavy, 2019-07-30:
Would arming still be prevented if there was no mag?  I wonder if it could be handled similar to other sensors like GPS and Airspeed where there are parameters to let users/integrators decide if the sensors are required for arming and other functions?

totomyl, 2019-07-31:
I am using the pixfalcon on 1.9.2, forgot to mention this

totomyl, 2019-08-06:
@julianoes Is it alright to run the V2 firmware on a pixfalcon? or should I be running some other version, I am not too sure of the difference between all the versions of the firmware (v2, v3, v4).

hamishwillee, 2019-08-06:
@totomyl Yes, but you should TRY to update to V3 firmware following the instructions here: http://docs.px4.io/master/en/config/firmware.html#bootloader

FMUv2 is very resource constrained, so quite a lot of drivers that you might want to use are not installed by default. If you can run FMUv3 (depends on version of the hardware) then you are less likely to run into problems. 

There is a little overview of the differences in FMU versions here: http://docs.px4.io/master/en/flight_controller/pixhawk_series.html#fmu-versions
The FMU version implies compatibility of firmware - so all FMUv2 boards can run the same firmware. An increased FMU version number only tells you that the firmware is not compatible with other FMU versions (e.g. they might be the same except for a minor connector difference). But _usually_ an increased FMU version number means "more capable hardware".

siddhu45, 2019-10-02:
I am not sure, if this issue is completely solved. I tried today with the latest master build of PX4 v1.10.0b and after reading through the #13023 and the related info, I made EKF2_AID_MASK to `24` (EV pos + EV yaw) , GPS_checks to `0`, EKF2_HGT_MODE to `Vision`. But the mag still interferes and there is a deviation from the external reference yaw causing the arming check error "Preflight Yaw estimate error". Do I need to set any other parameter? @julianoes any advice please!
Thanks

gyronikeleg, 2020-01-25:
Did this issue every get resolved? I am getting the same problem

bresch, 2020-05-13:
To disable mag fusion, use [EKF2_MAG_TYPE](https://dev.px4.io/master/en/advanced/parameter_reference.html#EKF2_MAG_TYPE) "none" (or "MC custom"), don't disable the mag completely.

closed 2020-05-13 by bresch