PX4/PX4-Autopilot #12579: AUAV-X2 AHRS drifting

Fpv4tw, 2019-07-30:
Hi,

I recently tried to use px4 on my older AUAV-X2 for a VTOL project and successfully compiled the latest (dev) fw and uploaded it to the FC. Then I started with airframe setup and sensor calibration. First gyro was calibrated without any issues and I continued with accelerometers. There, I just got the following errors over and over again, no matter what orientation of IMU selected: "Error : invalid orientation.". I then had a glance at the flight view with the artificial horizon and witnessed that the horizon has settled at around -178 degree roll though the FC has been laying around flat. Whenever I move the fc it drifts quite a lot and calibration of mag and acc sensors is impossible because of that. 

I also tried to flash a stable version but the behavior is the just the same. I'm kinda lost what to do. As a test if some sensors are dead I reinstalled Arduplane and verified if I can calibrate the sensors - which I could. That brings me to the conclusion that something either with the stock settings or the firmware may be wrong.

julianoes, 2019-07-30:
Sounds like an orientation of one of the sensors is off.

Do you have a console? In that case it would be helpful if you could share the output.

@dagar or @davids5 do you have this piece of hardware to check?

davids5, 2019-07-30:
@Fpv4tw - I may have this board.  So I can confirm I have the same version. Would be able to post a front and back side image of the HW you using?

Fpv4tw, 2019-07-30:
@julianoes: what do you mean by console? I compiled it via terminal (mac Osx). Which output are you looking for? I sure can try to provide you anything you're looking for tomorrow. 

Same is for @davids5 but I didn't use any peripheral devices like external mag etc, yet. Therefore it's just an usual, bare AUAV-X2 connected via USB. But yeah sure, I can send some pictures tomorrow evening of that helps.

Thank you both/all for having a look at this 😊

julianoes, 2019-07-31:
> @julianoes: what do you mean by console?

https://dev.px4.io/v1.9.0/en/debug/system_console.html

Fpv4tw, 2019-08-01:
I tried to setup the console but other than my FTDI getting very hot, no connection could be made though I followed the guide and used screen ( screen /dev/tty.AQSpider-SPPDev 57600 8N1 ).
Attached you'll find a picture of it in the 3d printed case wired up with the FTDI. 

I don't now if I put it the right but after flashing the latest stable version and calibrated the gyro (even before) the whole AHRS solution is going wild in the artificial horizon in all axis. It may stabilize itself somewhere but can be put off easily. 

[image]

Fpv4tw, 2019-08-04:
Any ideas? Are the connections right for the console right?

julianoes, 2019-08-05:
Are you sure that the wiring is correct. It shouldn't get "very hot" :thinking:

Fpv4tw, 2019-08-05:
I know it shouldn't but either way I put it (interestingly there are 2 different pictures for connections of the front). I figured the one where the micro SD slot is pictured on the right below is the correct one and hooked everything up as shown for serial 7 (=5). If I put it the other way (SD slot left) my pc gives a warning for too high current (short) and cuts it automatically. I only did it once though...should be enough 😅😜

Shall I try another serial or can I change settings somehow?

julianoes, 2019-08-05:
I don't know the AUAV-X2 and neither your FTDI, so I'm not sure what to say.

Fpv4tw, 2019-08-05:
I may try it again via windows and other serials later. Will post if that helped. Other than using the console, would it help if I have a look at the parameters in QGC? Is it possible to see in the parameters if an acc or mag is disabled or so?

Fpv4tw, 2019-08-06:
I tried other serials yesterday with no success, additionally I switched to windows but that didn't work out because the ftdi wasn't recognized. I tried others but these behaved just the same. 

Furthermore I played a little with the sensor orientation and recognized that the AHRS behaved normal when no rotation was selected. The drifting issue was still present though, even when flashing different versions (stable, beta, Master).

Fpv4tw, 2019-08-06:
Is it possible that the sensors aren't correctly linked to the hardware?
I'm just asking because the wiki site lists the following sensors:
Invensense MPU9250 9DOF
Invensense ICM-20608 6DOF
MEAS MS5611 barometer

Whereas the following are really in it:
ST Micro L3GD20 / ST Micro LSM303D / Invensense MPU 6000 
MEAS MS5611 barometer

I don't know where the Pixhawk version is linked to the sensors used but could this be a reason?

Fpv4tw, 2019-08-08:
@davids5 are you able to replicate the error? Or does somebody else have the auav-x2 in use with the current px4 firmware?

davids5, 2019-08-08:
@Fpv4tw - sorry I have not had the time. I do not what to test with the wrong hardware. How hard is it for you to open the case to take the picture?

Fpv4tw, 2019-08-08:
@davids5 it wasn't easy but I did it the day before yesterday. I can upload a picture this evening. Do you want me to take a picture of a special detail of just a general picture as it is an ordinary auav-x2.

davids5, 2019-08-08:
@Fpv4tw just detailed enough that I can see if I have the same board. Front and Back.

Fpv4tw, 2019-08-08:
@davids5 will do!

Fpv4tw, 2019-08-08:
[image]
[image]
@davids5 hope this helps

Fpv4tw, 2019-08-08:
[image]

Fpv4tw, 2019-08-12:
@davids5 can you confirm the board /issue?

davids5, 2019-08-12:
@Fpv4tw - Unfortunately I can not add much value here.  I do not have the same HW as you and the sensors would be different. 

FYI @pkocmoud

Fpv4tw, 2019-08-12:
@davids5 Thank you for having a look at this though ;)

Fpv4tw, 2019-08-24:
Any news on this or shall I just use APM 😅 would love to use px4 though

dagar, 2019-08-24:
@Fpv4tw can you share a log? https://logs.px4.io/

Set to log from boot (SDLOG_MODE = 2) with full sensor comparison (SDLOG_PROFILE Sensor comparison), then do something recognizable in the log we can compare.

For example you could...
 - start with the board level and facing north
 - rotate fully
 - point board straight up
 - rotate fully again

Then share the link to the log here and we can review it.

Fpv4tw, 2019-08-24:
@dagar sounds like a plan. Will do in the next few days and report back.

Fpv4tw, 2019-08-26:
@dagar hope this helps:
uploaded a fresh stable firmware and followed your instructions.
[https://review.px4.io/plot_app?log=c082ef0c-6703-45d2-9360-72de728c7e21](url)

Fpv4tw, 2019-09-30:
@dagar any news on this?

pkocmoud, 2019-09-30:
@Fpv4tw - If is was working and now is not I would assume there is a defective sensor. That said, this flight controller is obsolete. The X2 was never widely accepted and quickly replaced by the X2.1. I am happy to offer you a deeply discounted replacement X2.1 or X2.1-777 and I apologize for your inconvenience. Please PM me if you are interested.

@julianoes @dagar @davids5 - Thank you for your efforts and commitment to provide excellent support!

Fpv4tw, 2019-09-30:
@pkocmoud thanks for your offer! I don't know how to pm you, yet. Can you send me a pm?

It's odd though that it is fully normal when I use APM plane or so. No drift at all.

Thank you guys for your help!

pkocmoud, 2019-09-30:
@Fpv4tw Sure just email me at help @ mrobotics.io .

closed 2019-09-30 by pkocmoud