PX4/PX4-Autopilot #12590: Undetermined behaviors for a IRIS quadcopter in simulation after resetting !!

shrit, 2019-07-31:
**Describe the bug**
I am doing a simulation for trajectory generation for a set of IRIS quadcopters in Gazebo environment. I started a set of simulations, in which we generate trajectory for a quadcopter, and then we land the quadcopter and reset them to their original place in the simulator. 
Yesterday night around 2 am. The quadcopter, as described in the provided log, did not take off after resetting, instead it moved around in the same place and finally got flipped.

I do not know if it is related to the resetting the quadcopter early before the disarming or not.
I have also the precedent flights if it is necessary 

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=2c69527c-6ddd-4872-9b0b-e7f79dbf269b

julianoes, 2019-07-31:
Feel free to make a PR which enables resetting SITL.

shrit, 2019-07-31:
Where, Here or in SITL repository??

julianoes, 2019-07-31:
I think it needs support here.

shrit, 2019-07-31:
```
#ifndef _MODEL_RESET_HH_                                                                                                                                                                                          
#define _MODEL_RESET_HH_                                                                                                                                                                                          
                                                                                                                                                                                                                  
# include <gazebo/gazebo.hh>                                                                                                                                                                                      
# include <gazebo/physics/physics.hh>                                                                                                                                                                             
# include <gazebo/common/common.hh>                                                                                                                                                                               
# include <gazebo/msgs/msgs.hh>                                                                                                                                                                                   
                                                                                                                                                                                                                  
namespace gazebo                                                                                                                                                                                                  
{                                                                                                                                                                                                                 
  class ResetPlugin : public ModelPlugin                                                                                                                                                                          
  {                                                                                                                                                                                                               
                                                                                                                                                                                                                  
  public: ResetPlugin(){}                                                                                                                                                                                         
                                                                                                                                                                                                                  
  public: void Load(physics::ModelPtr _parent, sdf::ElementPtr /*_sdf*/)                                                                                                                                          
    {                                                                                                                                                                                                             
                                                                                                                                                                                                                  
      // Store the model pointer for convenience.                                                                                                                                                                 
      this->model = _parent;                                                                                                                                                                                      
                                                                                                                                                                                                                  
      // Create the node                                                                                                                                                                                          
      this->node = transport::NodePtr(new transport::Node());                                                                                                                                                     
                                                                                                                                                                                                                  
      this->node->Init(this->model->GetWorld()->Name());                                                                                                                                                          
                                                                                                                                                                                                                  
[... cut ...]
  GZ_REGISTER_MODEL_PLUGIN(ResetPlugin)                                                                                                                                                                           
}                                                                                                                                                                                                                 
                                                                                                                                                                                                                  
#endif       
```

shrit, 2019-07-31:
The idea, is that the user send msg that reset the quadcopters when it is needed

shrit, 2019-08-01:
@julianoes Any thoughts about this problem??

julianoes, 2019-08-05:
I think it needs support in PX4/Firmware. Either the px4 process needs to restart or at least the estimator needs to do a reset.

shrit, 2019-08-07:
@julianoes, Exactly, However, I thought about a semaphore, or somethings similar that need to be implemented in the MAVSDK that notify the user if the landing has successfully accomplished or not. In actual implementation the land and takeoff functions sends the user `Action::Result` about the command itself rather than the state of the drone. For example, I would happy if the land function is a blocking function until the quadcopter touch successfully the ground. What do you think?

julianoes, 2019-08-12:
> For example, I would happy if the land function is a blocking function until the quadcopter touch successfully the ground. What do you think?

You can already wait for that like this:

```
action->land();
while (telemetry->in_air()) {
    std::this_thread::sleep_for(std::chrono::seconds(1));
}
```

julianoes, 2019-11-11:
To summarize, resetting of a vehicle in SITL is currently not supported. If anyone wants to add a pull request that fixes that, please go ahead.

shrit, 2019-11-12:
@julianoes, I agree, I can create a pull request. However, resetting is more related to simulators rather than the SDK itself. 
A workaround in the form of a plugin can be realized, but we need to specify that we are in SITL rather than in real-life scenarios.
What do you think?

julianoes, 2019-11-13:
As I changed the title we don't have support in SITL/simulation. I don't see what MAVSDK could do about it.

shrit, 2019-11-13:
Ok, I understand, in this case, the pull request should be in SITL_Gazebo repository rather than in PX4

julianoes, 2019-11-13:
I think it will involve both: sitl_gazebo and PX4/Firmware.

shrit, 2019-12-16:
@julianoes, pull request done at https://github.com/PX4/sitl_gazebo/pull/380

shrit, 2020-03-15:
Keep open

shrit, 2020-06-13:
Keep open

Jaeyoung-Lim, 2020-06-13:
@shrit What is the status on https://github.com/PX4/sitl_gazebo/pull/380 ?

shrit, 2020-06-13:
It is pending for https://github.com/PX4/Firmware/pull/11654

shrit, 2020-09-12:
Keep open

Jaeyoung-Lim, 2021-01-13:
Closing since https://github.com/PX4/PX4-Autopilot/pull/11654 is not being continuded and this issue is stale.

Please reopen if otherwise

closed 2021-01-13 by Jaeyoung-Lim