PX4/PX4-Autopilot #12592: Here 2 GPS Not Detcted

UnmannedAirman, 2019-07-31:
Hey all,

The lab I work in utilizes roughly 40 Here 2 GPS units combined with Pixhawk 2.1s Cube Black PX4 firmware v3. Unfortunately, 15 of the Here 2s do not acquire satellites. The ones that do function correctly are using I2C and worked right out of the box. Interestingly enough, the Here 2s that do not work with PX4 are able to acquire satellites when using Ardupilot.  I've also posted on ProfiCNC's forums for additional help.

Any help would be appreciated. Thanks!

julianoes, 2019-08-05:
That's very odd, thanks for reporting.

Can you please get console output for the working and non-working case? And also the output of `gps status` for both?

waqaskhan200, 2019-08-20:
Hi,
I recently got 02 Here2 GPS units. I am trying to get them to work with Pixhawk 1. Fortunately, the Here2 comes with a cable to connect to Pixhawk 1 (GPS connector, I2C connector and switch connector).
When I flash the latest PX4 stable 1.9.2, the GPS doesn't work (0 satellites; see GPS not working image). I am also attaching the output of 'gps status'.
When I flash the latest Ardupilot stable, the GPS works fine.
[image]
[image]

julianoes, 2019-08-21:
So the GPS is connected to GPS port 1?

waqaskhan200, 2019-08-21:
Yes, on Pixhawk 1, it is the Gps port. I am going to get my hands on a cube today, i will try it on that today.

julianoes, 2019-08-21:
I can only suggest to debug it using a serial console or JTAG to find the root cause.

RomanBapst, 2019-08-21:
@julianoes @waqaskhan200 We just had the same issue with one of the newer here2 units.
Furthermore, we noticed that we cannot flash firmware to the newer batches of Pixhawk cubes.

RomanBapst, 2019-08-21:
@proficnc Are you aware of anything that changed? Currently we have two problems:
1) Newer here2 gps modules do not work with latest PX4. Older here2 modules work fine.
2) Newer Pixawk cubes cannot be updated via QGC nor do they talk mavlink through the usb interface.

UnmannedAirman, 2019-08-21:
Okay so I was able to update the firmware on the Here 2 GPS units to function in I2C on the Cubes. I'm not sure if this solution will work on the Pixhawk 1s but it's worth a try. 

1. Make sure you have Mission Planner 1.3.64. You can find the Mission Planner archives [here. ](http://firmware.ardupilot.org/Tools/MissionPlanner/archive/)

2. Download the firmware for the Cubes [here](http://firmware.ardupilot.org/Copter/latest/CubeBlack/). You will need to download the 'arducopter.apj' file and flash it to your Cube. You can do this by clicking 'Initial Setup' in Mission Planner and clicking 'Load Custom Firmware' located at the bottom right. 

3. After the Cube is flashed you'll need to update these parameters when connected to Mission Planner:

        CAN_D1_PROTOCOL：1
        CAN_D2_PROTOCOL：1
        CAN_P1_DRIVER：1
        CAN_P2_DRIVER：1
        GPS_TYPE：9
        NTF_LED_TYPES：231

4. After parameters have been written to the Cube, hard restart the Cube by unplugging the USB. 

5. Ensure the Here 2 GPS unit is plugged into the port labeled CAN 2 and the switch on the chip of the Here 2 is switched to CAN. 

6.. Connect to Mission Planner as normal via USB to the cube

7. In Mission Planner go to "Initial Setup" at the top. 
8. Click 'Optional Hardware' and then click "UAVCAN".

Your Mission Planner should now look like this:
[image]

9. Click SLCAN Mode CAN1 (this is labeled as Can2 on the carrier board)

10. Your Here 2 GPS should now display. 

[image]

11. Click 'Update' and then search the internet for the update. 

12. Mission Planner will download and install the latest CAN firmware for you.

[image]

13. At this point the LEDs should be noticeably different which is a sign that your device has been successfully updated. 

14. Once updated you can close out of Mission Planner and open up QGroundControl.

15. Ensure that you flip the switch on the Here 2 GPS back to I2C and plug in the serial cable connected to the Cube to 'GPS 1' on the board. 

Your Cube should now be attempting to acquire satellites in QGroundControl.

julianoes, 2019-08-22:
> 2. Newer Pixawk cubes cannot be updated via QGC nor do they talk mavlink through the usb interface.

Fixed in https://github.com/PX4/Firmware/pull/12772.

UnmannedAirman, 2019-08-22:
I flashed all my cubes using Mission Planner. You don't need to connect to MAVLINK to do this.

waqaskhan200, 2019-08-22:
After updating the Here2 units to the latest firmware as per the instructions of @UnmannedAirman, the GPS is working fine. @UnmannedAirman Thanks !!!
Is there a way to get a firmware file and a process to update the Here2 GPS without using the cube?
Thanks for the help guys.

sfuhrer, 2019-08-29:
We can't connect a Here2 with the following chip:

[image]

That's the error message I get when entering GPS status:
[image]

waqaskhan200, 2019-08-29:
@sfuhrer updating the Here2 firmware as mentioned in previous comments worked for us.

julianoes, 2019-11-12:
@proficnc do you know what could be different between these versions? Is there something we should try to talk to Here2 with older firmware or is updating the only solution?

mwiatt, 2019-11-14:
> Is there something we should try to talk to Here2 with older firmware or is updating the only solution?

@julianoes My brand new Here2 units wont talk to PX4. I dont think the problem is with the older firmware. I think the problem is with the newer firmware they are shipping with. I think that "updating" via MissionPlanner/Arducopter actually rolls back the firmware to one that is compatible with PX4. I am not 100% certain about this but my old Here2 worked with PX4 out of the box.

UnmannedAirman, 2019-11-14:
@mwiatt I believe you're right. I currently have over 50 Here 2 GPS units from different generations. In the past, they used to work just fine out of the box. Liked you said though, I think the firmware they are now shipped with is incompatible with the solution being to roll the GPS firmware back.

julianoes, 2019-11-15:
@mwiatt @UnmannedAirman ok thanks for the information! That's a bit concerning. @proficnc would be great to get your input on this!

mwiatt, 2019-11-15:
I think it would be generally useful if @proficnc would either communicate with the Dronecode foundation regarding changes to the hardware/firmware or test the changes against the current release of PX4/QGC before implementing them.

The Cube 2.1 bootloader was also changed on recent product that shipped which caused QGC to not be able to update firmware https://github.com/mavlink/qgroundcontrol/issues/7523

The @proficnc hardware and PX4/QGC software are used by a bunch of people in industry and the out-of-the-box experience should be smooth with current hardware and SW. I ended up spending 2 days searching around figuring out why I couldn't connect to my Cube and why my Here2 wouldn't talk to PX4. Since I had used older versions of these products without problems, it seemed like the HW was dead.

JacobCrabill, 2019-11-20:
@UnmannedAirman, @mwiatt have you guys tried using a [Zubax Babel](https://shop.titaneliteinc.com/index.php?route=product/product&product_id=1004) to see if the Here2 is configured properly?

I know that we aren't able to use Here2's out of the box either.  The solution for us has been to plug it into a Babel board, open the [UAVCAN GUI tool](https://uavcan.org/GUI_Tool/Overview/), then set a static ID.  After that, it should be plug & play via CAN with the Cube / PX4.  It appears that the dynamic node allocation functionality (by the UAVCAN driver within PX4) isn't able to set a UAVCAN node ID to the Here2?

mwiatt, 2019-11-20:
I haven't tried. I use the Here2 in UART/GPIO mode, not CAN, since I still want LEDs, arm switch, etc. I did try CAN mode when I was trying to figure out why UART mode wasn't working and it didnt work either.

With the current shipping firmware on the Here2, the Here2 wont talk to PX4 over CAN or UART.

julianoes, 2019-11-27:
I followed [these steps](https://docs.cubepilot.org/user-guides/here-2/updating-here-2-firmware) to update the Here2 and the GPS is now recognized correctly in the I2C/serial setting.

closed 2019-11-27 by julianoes