PX4/PX4-Autopilot #12594: Compass needs calibration after each airframe change

JMshare, 2019-07-31:
**Describe the bug**
Accelerometers and gyros calibration parameters are preserved, but the magnetometers calibration is wiped every time I change airframe. 

**To Reproduce**
Set for example the Generic Quadrotor x airframe and calibrate all sensors.
Change for example to DJI FlameWheel F450 airframe.
See that the compass calibration is needed again.

**Expected behavior**
I would expect the sensors calibration to stay preserved if I didn't change anything physically about the airframe, only redefined it in PX4.
The level horizon is also wiped, but it's easier to calibrate than the whole macarena with the compass so it's not such a big issue I think.

**System**
1.9.2 master

mhkabir, 2019-07-31:
If you're changing the airframe in PX4, how is the system supposed to know whether you actually changed the physical airframe? It is safer to wipe the mag calibration.

dagar, 2019-08-02:
If the airframe changes the the external magnetometer location, orientation, or even soft iron effects from the vehicle itself might have changed. Yes you can get away with it in various situations, but when it's not okay the problems can be subtle, so we err on conservative side.

dagar, 2019-08-02:
@JMshare I'm also wondering if/why you're regularly changing airframes?

closed 2019-08-02 by dagar