PX4/PX4-Autopilot #12598: Display nav_roll and nav_pitch in telemetry

moreba1, 2019-08-01:
Hi

I need to see navigation roll and pitch angle for tuning PID during flight but I can not find it. I think this parameters not send with Mavlink.

julianoes, 2019-08-05:
We need more information to help you.

- What angle exactly do you need to see?
- What ground station software are you using?
- What PX4 version are you using?
- What telemetry link are you using?

moreba1, 2019-08-05:
I need to see navigation angle of roll and pitch . i use QGroundcontrol and latest px4 v1.9.2 . for telemetry i use 3dr telemetry or RFD900 .

julianoes, 2019-08-05:
So you mean you need to see the setpoint of the roll and pitch angle?

Antiheavy, 2019-08-05:
I suspect the request is to be able to see both the roll/pitch setpoints and also actual roll/pitch values and display them live (using the same units and scale) in the QGC Analyze Widget for purposes of tuning.  I recall being able to do this with older versions of PX4, but with more recent releases of PX4 I too have been unable to figure out if both of these values (setpoints and actual) are still able to be displayed by the widget.

julianoes, 2019-08-05:
Ok, got it. I wonder if QGC has the functionality to request custom messages at a certain rates.

What you could try is to add an extras.txt as described [here](https://dev.px4.io/master/en/concept/system_startup.html#customizing-the-system-startup) with this line:

```
mavlink stream -r 50 -s ATTITUDE_TARGET -u /dev/ttyS1
```

I'm not sure if it is `/dev/ttyS1`, this depends on the hardware that you're using and wiring setup.

moreba1, 2019-08-06:
[image]

currently we can see roll and pitch angle in QGC Analyze Widget but they are not in degree.

julianoes, 2019-08-12:
>  currently we can see roll and pitch angle only in QGC Analyze Widget but they are not in degree.

Are they in radians?

moreba1, 2019-08-14:
yes , they are radians.

julianoes, 2019-08-15:
What about the field `roll sp` on the screenshot above? Isn't that what you want together with `ATTITUDE.roll`? It's both in rad but at least both have the same unit.

moreba1, 2019-08-16:
"roll sp" is roll rate . i noticed during flight roll set point appears (in degree). so just roll angle change to degree .

[image]

julianoes, 2019-08-16:
> "roll sp" is roll rate . 

Really? I don't think so:
https://github.com/mavlink/qgroundcontrol/blob/4b6163ecfd35aeefa45391dcc477122177bea50a/src/uas/UAS.cc#L301

moreba1, 2019-08-16:
i was wrong . it is roll setpoint ( rad)

closed 2019-08-19 by julianoes