PX4/PX4-Autopilot #12610: Rally points seem to be ignored in 1.9.0 and 1.9.2

RyanHurst, 2019-08-02:
**Describe the bug**
Rally points seem to be ignored in 1.9.0 and 1.9.2. Tested with Teal one and in gazebo simulator.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on
2. Uploaded some rally points
3. Fly past rally points, and command drone to return to home.
4. See error

**Expected behavior**
I would expect the drone to return to the rally point instead of the home position. When exactly is the drone supposed to return to a rally point instead of to the home position? Is it just whichever is closer or...

Add screenshots to help explain your problem.
[image]

**Drone (please complete the following information):**
- Teal One. Iris gazebo simulation

dagar, 2019-08-02:
FYI @bkueng

hamishwillee, 2019-08-04:
I didn't think this had been implemented in PX4. Certainly there is no documentation under the terms "Rally point" or "Safe landing point", and looking at the code I can see that the values are stored, but not that they are used. 

@bkueng If this does exist can you confirm so I can document.

bkueng, 2019-08-05:
> I didn't think this had been implemented in PX4.

Correct.

@RyanHurst you're welcome to add support for it.

hamishwillee, 2019-08-06:
@bkueng @RyanHurst In the meantime, might it be worth adding a note to explicitly that we don't support rally points? (we don't normally put "we don't support this" for unsupported things)

Problem is that I'm not sure where. How about in the [RTL/Return docs here](http://docs.px4.io/master/en/flight_modes/return.html)?
 
On a similar note, we don't say much about geofence. Do you think worth creating as a peer of [missions](http://docs.px4.io/master/en/flying/missions.html) ?

bkueng, 2019-08-06:
> @bkueng @RyanHurst In the meantime, might it be worth adding a note to explicitly that we don't support rally points?

We can do that.

> On a similar note, we don't say much about geofence. Do you think worth creating as a peer of missions ?

I think geofence deserves its own page (you can enable it in other modes as well).

julianoes, 2019-11-04:
Rally points have been added in #12696.

closed 2019-11-04 by julianoes