PX4/PX4-Autopilot #12613: Posix rc file overrides MAVLink rates

pgorczak, 2019-08-03:
**Describe the bug**
The rc file for the POSIX target contains [`mavlink stream` commands](https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/rcS#L238-L246) that override defaults.

**Expected behavior**
I expected the message types and rates from MAVLink mode "normal" to apply when running a SITL simulation. This is also suggested by the console output

    INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550

julianoes, 2019-08-05:
@pgorczak thanks for raising this. That's indeed not super nice. Would you be interested to create a pull request to fix this?

pgorczak, 2019-08-09:
Sure! Pull request is at  #12672

julianoes, 2019-08-12:
Thanks @pgorczak