PX4/PX4-Autopilot #12621: build px4_fmu-v5_rtps fails at generate_microRTPS_bridge.py using docker image px4-dev-nuttx:latest

tschiu, 2019-08-04:
**Describe the bug**
make px4_fmu-v3_rtps
make px4_fmu-v4_rtps
make px4_fmu-v5 rtps
all fail with the following error

Traceback (most recent call last):
  File "/src/Firmware/msg/tools/generate_microRTPS_bridge.py", line 401, in <module>
    generate_agent(agent_out_dir)
  File "/src/Firmware/msg/tools/generate_microRTPS_bridge.py", line 334, in generate_agent
    "fastrtpsgen not found. Specify the location of fastrtpsgen with the -f flag")
Exception: fastrtpsgen not found. Specify the location of fastrtpsgen with the -f flag

**To Reproduce**
I am running a fresh Ubuntu 16.04 Server VM with Apache2 and MySQL only installed (it's a clone of my standard LAMP server).  Server is still on 16.04 and hasn't been upgraded to 18.04 yet.

Install docker using script found here https://dev.px4.io/v1.9.0/en/test_and_ci/docker.html
# Exact script below:
curl -fsSL get.docker.com -o get-docker.sh
sudo sh get-docker.sh

Added me to docker group:
# Create docker group (may not be required)
sudo groupadd docker
# Add your user to the docker group.
sudo usermod -aG docker $USER
# Log in/out again before using docker!
logout
...
login with bash

# get dev-nuttx image
docker pull px4io/px4-dev-nuttx

# get PX4 source
cd ~
mkdir src
cd src
git clone https://github.com/PX4/Firmware.git

# run the container with bash shell
docker run -it --privileged --env=LOCAL_USER_ID="$(id -u)" -v ~/src/Firmware:/src/Firmware:rw -v /tmp/.X11-unix:/tmp/.X11-unix:ro -e DISPLAY=:0 -p 14556:14556/udp --name=px4nuttx px4io/px4-dev-nuttx:latest bash

# docker container created and are in bash shell
user@6c4a4043cbca:/#

# these builds work
make px4_fmu-v3
make px4_fmu-v4
make px4_fmu-v5

# these fail
make px4_fmu-v3_rtps
make px4_fmu-v4_rtps
make px4_fmu-v5_rtps

Although the v3_rtps does succeed when I'm running local on the VM after installing using the convenience scripts provided.

**Expected behavior**
1.  Consistent builds (the docker image compile should compile the build versions as my local install -- specifically, the v3_rtps compiles on local, doesn't on docker container)
2.  rtps feature should compile on all configurations.

**Log Files and Screenshots**
user@6c4a4043cbca:/src/Firmware$ make px4_fmu-v5_rtps
-- PX4 config file: /src/Firmware/boards/px4/fmu-v5/rtps.cmake
-- PX4 config: px4_fmu-v5_rtps
-- PX4 platform: nuttx
-- PX4 lockstep: disabled
-- PX4 version: v1.9.2
-- cmake build type: MinSizeRel
[... 62 lines cut ...]
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/lib/ccache/arm-none-eabi-gcc
-- Check for working C compiler: /usr/lib/ccache/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.15")
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1
-- Configuring done
-- Generating done
-- Build files have been written to: /src/Firmware/build/px4_fmu-v5_rtps/external/Build/px4io_firmware
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'px4_fmu-v5_rtps' failed
make: *** [px4_fmu-v5_rtps] Error 1

**Drone (please complete the following information):**
N/A

**Additional context**
I *think* this is about as plain-vanilla as you can get.
To get xhost to work, I needed to install x11-server-utils
sudo apt-get install x11-xserver-utils

TSC21, 2019-08-04:
@tschiu have you tried using `px4-dev-nuttx:2019-07-29` instead?

tschiu, 2019-08-04:
No.  Will do so right now

tschiu, 2019-08-04:
#THANK YOU!!!!!  

That works perfectly.  

I've been pulling my hair out for a week trying to get this to compile.

v3, v4, v5 and v3_rtps, v4_rtps, v5_rtps all compile without errors.

Question:  I gather pulling "latest" docker images doesn't guarantee success, and that, if I need to update the docker image, I should use the most recent version with a date tag?

tschiu, 2019-08-04:
Using pxio/px4-dev-nuttx:2019-07-29 instead of pxio/px4-dev-nuttx:latest solved the issue.

closed 2019-08-04 by tschiu