PX4/PX4-Autopilot #12631: Unable to add new airframe for FMU-v5

themcfadden, 2019-08-05:
**Describe the bug**
Following the instructions outlined in https://dev.px4.io/master/en/airframes/adding_a_new_frame.html, I am unable to add a new airframe under FMU-v5. Everything builds with no issues, but nothing shows up in QGC and the airframes.xml file in the build directory is not updated. 

I am able to successfully do it under the FMU-v4 build.

A couple notes:
- The 4250_Teal, and others are not showing up in my fmu-v5 build. The airframes are there and listed in CMakeLists.txt, but are not part of the build output.
- Even commenting out something in airframes/CMakeLists.txt does not change anything.
- I believe the official FW release has the Teal and other missing airframes.

**To Reproduce**
Steps to reproduce the behavior:
1. Update or create a new airframe in Firmware/ROMFS/px4fmu_common/init.d/airframes/ as per instructions pointed to above.
2. make px4_fmu-v5
3. Check Firmware/build/px4_fmu-v4_default/airframes.xml and observe no changes.

**Expected behavior**
Airframe is created as in other targets.

**Log Files and Screenshots**
N/A

**Drone (please complete the following information):**
N/A but I'm working on a Teal drone

**Additional context**

hamishwillee, 2019-08-05:
I haven't tried this, but in step 3, shouldn't you be checking the v5 path? `Firmware/build/px4_fmu-v5_default/airframes.xml`

themcfadden, 2019-08-05:
> I haven't tried this, but in step 3, shouldn't you be checking the v5 path? `Firmware/build/px4_fmu-v5_default/airframes.xml`

Yes, correct. That is a cut and paste error.

AlexandreBorowczyk, 2019-08-06:
Are you sure the ID you attributed to the new config is unused?

julianoes, 2019-08-06:
And you don't have an `exclude` like here?

https://github.com/PX4/Firmware/blob/cfba41d2f47f248871e6c660a790686161ec14bf/ROMFS/px4fmu_common/init.d/airframes/4250_teal#L11

RicardoM17, 2019-08-06:
Check what @julianoes just said. I literally just copied the 4250_teal airframe file into a 4251_teal, deleted that line and it's building on v5.
[image]

themcfadden, 2019-08-06:
Yep, that's it. Thanks for the quick response!

Just curious, how is the airframe.xml file generated? Is there a script I can review?

mhkabir, 2019-08-06:
https://github.com/PX4/Firmware/tree/master/Tools/px4airframes

Closing as resolved.

closed 2019-08-06 by mhkabir

RicardoM17, 2019-08-07:
@themcfadden I believe this is what you are looking for?

https://github.com/PX4/Firmware/blob/master/Tools/px_process_airframes.py