PX4/PX4-Autopilot #12637: Simple Mixer output Scaler doesn't work

AlexandreBorowczyk, 2019-08-05:
**Describe the bug**
Adding a output scaler line (e.i.: O: -scale +scale offset min max) to a simple mixer was no effect.

**To Reproduce**
1. Build and Flash https://github.com/AlexandreBorowczyk/Firmware/tree/bug/output_scaler
2. Select X-UAV mini Talon configuration
3. Arm 
4. Observe Pwm output of channel 1 

**Expected behavior**
Output of channel should have a offset.

**Log Files and Screenshots**
Mixer is load correctly:
```
INFO  [px4io] default PWM output device
INFO  [init] Mixer: /etc/mixers/AAVVTWFF_vtail.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
ERROR [pwm] PWM_SERVO_SET_SELECT_UPDATE_RATE
```
PWM output is not offset
```
pwm info
device: /dev/pwm_output0
channel 1: 1500 us (default rate: 50 Hz failsafe: 1560, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 2: 1500 us (default rate: 50 Hz failsafe: 1500, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 3: 1500 us (default rate: 50 Hz failsafe: 1500, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 4: 1500 us (default rate: 50 Hz failsafe: 1500, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 5: 1000 us (default rate: 50 Hz failsafe: 900, disarmed: 1000 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 6: 1500 us (default rate: 50 Hz failsafe: 1500, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 7: 1125 us (default rate: 50 Hz failsafe: 900, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel 8: 1875 us (default rate: 50 Hz failsafe: 2100, disarmed: 0 us, min: 1000 us, max: 2000 us, trim:  0.00)
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4
```

julianoes, 2019-08-06:
@RomanBapst do you have experience with this?

AlexandreBorowczyk, 2019-08-06:
I've tried checking out  the Stable Branch then set
`#define debug(fmt, args...)	do { printf("[mixer] " fmt "\n", ##args); } while(0)`
in mixer.cpp, mixer_simple.cpp and mixer_group.cpp
Here is the output, which is really unexpected...
It fails parsing but still manage to create a functioning mixer!?!
```
[mixer_simple] control parse failed on 'S: 0 6 10000 10000'
[mixer_simple] simple mixer parser failed parsing ctrl scaler tag, ret: 'S: 0 0 -10000 -10000 0 -10000 10000
S: 0 6 10000 10000'
[mixer_simple] loaded mixer with 2 input(s)
[mixer_group] SUCCESS - buflen: 43
[mixer_simple] control parser failed finding tag, ret: 'S:'
[mixer_simple] simple mixer parser failed parsing ctrl scaler tag, ret: 'S: 0 0 -10000 -10000 0 -10000 10000
S:'
[mixer_simple] loaded mixer with 2 input(s)
[mixer_group] SUCCESS - buflen: 25
[mixer_simple] control parse failed on 'S: 0 2 7000 7000 0 -'
[mixer_simple] simple mixer parser failed parsing ctrl scaler tag, ret: 'S: 0 2 7000 7000 0 -'
[mixer_simple] loaded mixer with 2 input(s)
[mixer_group] SUCCESS - buflen: 13
[mixer_simple] control parser failed finding tag, ret: 'S: 0 2 7'
[mixer_simple] simple mixer parser failed parsing ctrl scaler tag, ret: 'S: 0 2 7'
[mixer_simple] loaded mixer with 2 input(s)
[mixer_group] SUCCESS - buflen: 3
[mixer] pre-parser rejected: No newline in buf
[mixer_simple] loaded mixer with 1 input(s)
[mixer_group] SUCCESS - buflen: 22
[mixer_simple] control parse failed on 'S: 0 2 -10000 -10'
[mixer_simple] simple mixer parser failed parsing ctrl scaler tag, ret: 'S: 0 2 -10000 -10'
[mixer_simple] loaded mixer with 1 input(s)
[mixer_group] SUCCESS - buflen: 40
[mixer_simple] loaded mixer with 1 input(s)
[mixer_group] SUCCESS - buflen: 1
[mixer] pre-parser rejected: No newline in buf
[mixer_simple] loaded mixer with 1 input(s)
[mixer_group] SUCCESS - buflen: 1
```

farzadsw, 2021-12-06:
I recently faced the same problem.
when I define a simple mixer, If I use the “O:” tag, the offset doesn’t have any effect.
for example:
M: 3
O: 10000 10000 -9000 -10000 10000
S: 0 0 -1000 0 0 -10000 10000
S: 0 1 -1000 0 0 -10000 10000
S: 0 2 -1000 0 0 -10000 10000

in this case, the -9000 (offset) is not applied to the output.
After checking the source code and some debugging, I found out that in the PWMOut.cpp file the software reads the values from PWM_MAIN_TRIMx (0 by default) and replaces the output offset (-9000) from the mixer file!

_mixing_output.mixers()->set_trims(values, _num_outputs);
in other words, the output offset in the mixer file never gets used.

I don’t think the PWM_MAIN_TRIMx parameter is a good replacement for the output offset in the mixer file (if it is at least they should mention it in the docs or remove it from the mixer altogether).

bkueng, 2022-10-25:
Mixers got removed (https://github.com/PX4/PX4-Autopilot/pull/19990)

closed 2022-10-25 by bkueng