PX4/PX4-Autopilot #12654: ERROR [mavlink] Failed to start shell (-l) in PX4 Gazebo Sim

bozkurthan, 2019-08-07:
**Describe the bug**
This problem occurs on PX4 and QGC Latest build in Ubuntu 16.04 | 18.04. 
PX4 Gazebo Simulation environment couldn't start mavlink shell. When I try to send via QGroundControl Mavlink Console to PX4 stack it produces "Failed to start shell (-1) "
I think it is related to UDP address.

**To Reproduce**
Steps to reproduce the behavior:
1. Simulation is started with `make posix gazebo`
2. QGrouldControl is started and tried to send Mavlink command from Console
3. Mavlink console publishes `ERROR [mavlink] Failed to start shell (-l)`
4. See error

**Expected behavior**
It can be able to send command from Mavlink Console to PX4 stack.

From PX4 shell mavlink status:
```
pxh> mavlink status

instance #0:
	GCS heartbeat:	208001 us ago
	mavlink chan: #0
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 10.924 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 1.000
	  tx rate max: 4000000 B/s
	  rx: 0.021 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Normal
	MAVLink version: 2
	transport protocol: UDP (14570, remote port: 14550)
	partner IP: 127.0.0.1
	ping statistics:
	  last: 0.00 ms
	  mean: 0.00 ms
	  max: 0.00 ms
	  min: 0.00 ms
	  dropped packets: 0

instance #1:
	mavlink chan: #1
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 15.082 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 1.000
	  tx rate max: 4000000 B/s
	  rx: 0.000 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Onboard
	MAVLink version: 1
	transport protocol: UDP (14580, remote port: 14540)
```

Add screenshots to help explain your problem.

Screenshot:
[image]

Console output:

```
hanco@new-host-name:~/GCS_Tasarim/Firmware$ sudo make posix gazebo
[2/6] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++17 compiler
[... 130 lines cut ...]
[Msg] Publicized address: 192.168.42.62
[Wrn] [GuiIface.cc:122] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
INFO  [px4_work_queue] creating: wq:hp_default, priority: 87, stack: 16384 bytes
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-08-07/06_55_56.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [mavlink] partner IP: 127.0.0.1

pxh> 
pxh> INFO  [ecl/EKF] EKF commencing GPS fusion
ERROR [mavlink] Failed to start shell (-1)
ERROR [mavlink] Failed to start shell (-1)
ERROR [mavlink] Failed to start shell (-1)
ERROR [mavlink] Failed to start shell (-1)
ERROR [mavlink] Failed to start shell (-1)
ERROR [mavlink] Failed to start shell (-1)

```

bkueng, 2019-08-07:
There's no support for that in SITL.

closed 2019-08-07 by bkueng

junwoo091400, 2022-06-02:
This is not supported because the QGC MAVLink Shell accesses the "NuttX shell", which doesn't exist in SITL since it runs on top of POSIX (your local machine)!

However, you can access individual commands / modules of the PX4 instance you are running via accessing the binary files located in the "build/px4_sitl_default/bin" folder. Note however that all the commands are prefixed with `px4-`, so you need to include that to execute the commands!

Here, I executed the command `./build/px4_sitl_default/bin/px4-uorb top`, which shows the output of the equivalent command `uorb top` inside the px4 shell.
[image]

More info is available here: https://docs.px4.io/master/en/concept/system_startup.html#posix-linux-macos