PX4/PX4-Autopilot #12659: Internal temperature sensor.

Louisdirk, 2019-08-08:
**Describe the bug**
In order to get the MCU internal temperature, I upgraded the Pixharwak Pro V1.3 to the v1.9.2 Firmware. But the temperature seems incorrect.

**To Reproduce**
Steps to reproduce the behavior:
1. listener adc_report
[image]
The last channel value is Vsense = 0.0032V.

2. calculate temperature value according to datasheet:
[image]
[image]

Temperature = (0.0032-0.76)/0.0025+25 = -277.72℃

Is it corret? Could you help to check? @dagar

Louisdirk, 2019-08-20:
@davids5  Could you have a look?

julianoes, 2019-08-20:
@Louisdirk looks like everyone is busy. I suggest you dive into the code yourself and check. A pull request with a fix would be much appreciated.