PX4/PX4-Autopilot #12676: scaling airspeed value MS4525DO

moreba1, 2019-08-09:
Hi 

I used MS4525DO airspeed sensor with pixhawk cube , i calibrated sensor but it read airspeed not ture. in calibration just zero offset adjusted but also there is need to a scale parameter . 

[image]

sfuhrer, 2019-08-10:
hi @moreba1 

This looks to me like your sensor has major issues. Is it maybe partially clogged? 

Concerning an aiirspeed scaling factor: Such a thing has yesterday been merged into master with https://github.com/PX4/Firmware/pull/12353. New there is a parameter "ARSP_ARSP_SCALE" for that, or it can be estimated automatically. Note though that currently the scaled airspeed is not used in the control modules yet (but still the old 'normal', unscaled airspeed).

For you a scaling anyway wouldn't be the solution, as your reading doesn't really correlate with the actual airspeed but rather just stays constant..

moreba1, 2019-08-10:
sensor work properly and pitot tube is opened. I tested in a windy day and i think scaling factor of my sensor is about 2 .

sfuhrer, 2019-08-13:
> sensor work properly and pitot tube is opened. I tested in a windy day and i think scaling factor of my sensor is about 2 .

Okay, if it was windy than maybe it really was just the scaling. Have you tried flying with #12353 / current master? The you could set ARSP_ARSP_SALE to 2 and see whether then the airspeed_validated.true_arispeed_m_s makes sense. 
To estimate the scale factor more precisely, enable online estimation of it via setting of ARSP_SCALE_EST to 1 (setting it to 0 again will save the current scale estimate). Let me know how it works if you do it, and I'd be happy to help you further.

I'm closing this issue as we already have your requested feature now.

closed 2019-08-13 by sfuhrer