PX4/PX4-Autopilot #12685: Arm authorization fails on first attempt, succeeds erroneously after

mattagiii, 2019-08-13:
I’m trying to get arm authorization working and running into some issues. First here’s my setup:

- Pixhawk Cube
- Authorizer is a Linux board running pymavlink using v2 dialect
- Param COM_ARM_AUTH = 1280001 (require one arm, 5s timeout, authorizer sysid 1)
- Param COM_MIS_REQ = 2 (require auth but not valid mission)
- Arming via Qground

I can go into more detail on the setup if needed, but all the basics appear to be working. When arming I see the COMMAND_LONG 3001 arrive in the python script, and I have even rebuilt PX4 for the Cube to check that the COMMAND_ACKs I send to it are being received in the mavlink module. Here is the behavior I observe:

- After reboot of Cube, the first arming attempt fails. It appears to fail even before the first COMMAND_ACK (with result MAV_RESULT_IN_PROGRESS) reaches the arm_auth module via uORB
- Subsequent arming attempts succeed, but they appear to rely on acks that are still queued from the first attempt. I have tested killing the python script after the first arming attempt (fail) and the second attempt still succeeds, despite no new acks being sent.

Here is a snippet showing what the python script does:
```
in_msg = self._mav_sock.recv_match(blocking=False)
if in_msg != None:
if in_msg.get_type() == ‘COMMAND_LONG’:
self._mav.command_ack_send(3001, 5, 255, 5, 1, 1) # in progress
self._mav.command_ack_send(3001, 0, 255, 5, 1, 1) # arm cmd, accept
```

A log probably isn't that helpful in this scenario but here it is anyway:
https://logs.px4.io/plot_app?log=bb57f1ee-6991-4190-b62b-6d2bdc7d013d

Here's the nsh listener output showing that auth fails before my acks arrive (on this run I rebuilt PX4 with extra debug messages):
[arm_auth_debug_fail_on_first_auth_attempt_txt.txt](https://github.com/PX4/Firmware/files/3495153/arm_auth_debug_fail_on_first_auth_attempt_txt.txt)

Please let me know what additional info would help.

This was also posted on the forum here: https://discuss.px4.io/t/arm-authorization/12383

julianoes, 2019-08-13:
`COM_MIS_REQ` -> `COM_ARM_MIS_REQ`, right?

julianoes, 2019-08-13:
I can reproduce this issue. I'm looking into it.

mattagiii, 2019-08-13:
@julianoes yes, correct that was a typo. Glad to hear you can reproduce it. Thanks, and let me know if you need any more info!

julianoes, 2019-08-13:
What I found until now in SITL is that the first vehicle_command is published in commander but not received and sent in the mavlink module. On subsequent arming tries the vehicle command is sent successfully in the mavlink module. I did not see this behaviour in `stable`/1.9.2, however, I could still see the immediate "Arm auth: No response".

I don't know yet if the issue I see is related to the one you have described.

@dagar any chance the first vehicle_command can get dropped after the uORB changes?

julianoes, 2019-08-13:
@mattagiii thanks for the detailed infos, that was very helpful!

julianoes, 2019-08-14:
@mattagiii can you try again with latest master? It looks like it was fixed in https://github.com/PX4/Firmware/pull/12674.

julianoes, 2019-08-14:
@mattagiii please try https://github.com/PX4/Firmware/pull/12698. I think it should resolve the initial fail, however, subsequent armings from leftover acks might still be happening. I'm looking into that now.

closed 2019-08-14 by dagar

mattagiii, 2019-08-15:
@julianoes your changes appear to work! Thanks!

With regard to the leftover acks, I think because your fix stopped multiple command_longs from being sent, there can't be more than one ack published? I tried sending extra acks and wasn't able to cause an arm. But at the same time I may not be qualified to make the call and say it's correct now.

If you think that problem can't occur anymore, then everything is resolved!

julianoes, 2019-08-15:
Yes, I think it is solved. I had added an extra commit to prevent retransmissions of commands.
https://github.com/PX4/Firmware/pull/12698/commits/49960a55b73f27a79979cea02d2a5a397325efac

PR #12698 (2019-08-14): commander: fix arming auth param translation
The int32 param COM_ARM_AUTH is mapped to a packed struct. However, this struct was not actually packed (anymore) and therefore the values were applied incorrectly.

I fixed this by applying the packed attribute. By using a union with a int32_t I could rid of the warning about address-of-packed-member.

This also fixes the issue where sysid and compid in the mavlink_command_sender do not match because the target_system and target_component are 0 (to all). This meant that the command was being re-transmitted multiple times even though an ack was rece [... cut]