PX4/PX4-Autopilot #12688: Commander mode initialization

MaEtUgR, 2019-08-13:
**Describe problem solved by the proposed feature**
I think one crucial point of the commander rewrite (https://github.com/PX4/Firmware/issues/7055) is that the state machine doesn't just initialize in manual mode and the user can by accident arm in manual mode even if the vehicle/end-product doesn't officially support that mode. This is a huge problem and even less experienced hobby users or developers struggle with this logic and do mistakes.

**Describe your preferred solution**
As quickly discussed with @julianoes a class handling the intention of the user regarding the mode in terms of configuration, different inputs, switching would be useful. The output of that would then just be what the user wants decoupled from any feasibility, failsafe, execution. The state machine should then start in an initialization state that cannot by accident be used to start the vehicle and from there the class representing the user intention can start switching into the appropriate mode. Depending on the use case the defaults can then be configured such that a production vehicle might never default to manual mode or even support it as the user intention.

Medium-term I'd like to try to come up with a useful proposal such that we can do some progress and iterate without rewriting everything at once. The goal is to relieve the main commander module from some functionality while improving the user experience and improve test coverage. Hopefully I'll find the time for that.

MaEtUgR, 2020-02-19:
This is a reoccuring problem and we keep on coming up with workarounds, I think it's worth a more clean solution. I hope I get to it soon.

MaEtUgR, 2021-03-10:
Related: https://github.com/PX4/PX4-Autopilot/pull/16614

bkueng, 2022-10-25:
Done in https://github.com/PX4/PX4-Autopilot/pull/20172

closed 2022-10-25 by bkueng