PX4/PX4-Autopilot #12690: Distance senslr ll40ls will never be executed without running IMU driver

hyonlim, 2019-08-13:
**Describe the bug**
Intentionally, we have tested following case.
1. Do not execute any IMU driver app (e.g., mpu9250 or so).
2. Execute "ll40ls" as the only sensor app. ( nsh> ll40ls start )
3. ll40ls will never be executed.

**To Reproduce**
As above.

**Expected behavior**
Clearly, ll40ls should work.

mcsauder, 2019-08-13:
Hi @hyonlim , a couple of suggestions to help initially.

1) Ensure that your SENS_EN_LL40LS parameter is set appropriately, (and if not, set/save the value and reboot your autopilot):
[image]

2) This could be related to issue #12278.  I am doing some work refactoring this driver to try to fix this issue, but my progress has been slowed by other priorities recently.  I'll keep up on this issue and work with you to resolve it.

dagar, 2019-11-11:
I believe this was resolved by increasing the maximum number of tasks.

closed 2019-11-11 by dagar