PX4/PX4-Autopilot #12692: Pixhawk Cube_calibration issue on Master

Tony3dr, 2019-08-13:
**Describe the bug**
When setting up the vehicle for a flight, on arming the vehicle I get a "Compass not found" message, see image below:
[image]
This issue is on Master e25db016 & PR Pull[12650](https://github.com/PX4/Firmware/pull/12650#event-2540347567), tried on Stable version and was able to fly. v1.9.2 (10690587).
**To Reproduce**
Steps to reproduce the behavior:
1. Power on the Drone
2. Uploaded mission 
3. Arm vehicle and Takeoff
4. Error, "Preflight Fail: COmpass #2 uncalibrated"
5. Calibrate vehicle 
6. Error message Gyro 1 no matching uORB devid
7. Unable to calibrate the vehicle on Master
[image]

**Log Files**
Compass not found and Calibration issue log on Master:
https://review.px4.io/plot_app?log=13cc9db5-4a03-41ff-a9e2-b80013d3e104

Stable  v1.9.2 (10690587) log:
https://logs.px4.io/plot_app?log=71a782cc-7979-4ff1-a6e8-f3d8f4757f55

PR 12650 log issue:
https://review.px4.io/plot_app?log=18094d44-f808-4b67-b7d6-8ef0df7f4f05
 
**Drone **
Frame- DJI F450
Flight Controller: Pixhawk 2(The Cube)

dagar, 2019-08-14:
I believe this is a bug with the icm20948 driver, which up until recently was using an incorrect device id. I'll try to reproduce.

proficnc, 2019-10-26:
So was this introduced bug fixed?

darshanz, 2019-11-13:
Is it fixed yet?? facing same problem.

FlavioTonelli, 2019-12-06:
I'm facing a similar issue with Here2 compass on a pixhawk 4 board.
That seems to confirm the bug is related to ICM20948 driver.

JacobCrabill, 2019-12-12:
Is this pull request relevant?
https://github.com/PX4/Firmware/pull/11189

feemi, 2019-12-16:
This issue still exists on pixhawk cube 2.1 fmu-v3 autopilot.

FlavioTonelli, 2020-01-10:
I can't reproduce anymore this issue on current stable version.
Tested on both Pixhawk v3 and v5 autopilot, equipped with an Here 2 antenna.
Is that been fixed?

JacobCrabill, 2020-03-24:
It's possible that this has been fixed by #14345, if the issue was in regards to external UAVCAN compasses.

mrpollo, 2020-04-06:
@Tony3dr Can you please verify @JacobCrabill comment? it would be great to resolve/close this issue asap.

LorenzMeier, 2021-06-02:
Solved a long time ago.

closed 2021-06-02 by LorenzMeier