PX4/PX4-Autopilot #12716: Throttle goes up to 100 % and over FW_THR_MAX

moreba1, 2019-08-16:
Hi

In climb there is a wrong happening. Throttle goes up to 100 % and over FW_THR_MAX will cause overheating and damage motor . 

[image]

julianoes, 2019-08-16:
Could you share a link to the log please?

moreba1, 2019-08-16:
yes ,

https://review.px4.io/plot_app?log=b6146ff1-7a47-4259-a6ed-be9f05e1d172

dsaffo, 2019-08-17:
Could this be a problem with your altitude set points? Looks like its still trying to be at a 1,000 meters when its trying to land. 

[image]

Thrust is encoded on the flight path (black low -> red high) and the charts on the right are filtered to only show data after the auto-return command (red square around 31 seconds on the timeline). Charts on the left showing thrust, alt set point, estimated and recorded  in that order.

RomanBapst, 2019-08-18:
@moreba1 This is why it happens:
https://github.com/PX4/ecl/blob/62fa464e4dd6cb5e585d23bd353aa50a469879f7/tecs/tecs.cpp#L319

TECS was in an underspeed conditions for all the time when the throttle values was at 100%.

RomanBapst, 2019-08-18:
[image]

moreba1, 2019-08-18:
@RomanBapst 
The motor had a problem with this test .  But the thrust should not exceed the FW_THR_MAX under any conditions.

sfuhrer, 2021-01-19:
> @RomanBapst
> The motor had a problem with this test . But the thrust should not exceed the FW_THR_MAX under any conditions.

Let's close this one and try to bring https://github.com/PX4/PX4-Autopilot/pull/15321 to a resolution.

closed 2021-01-19 by sfuhrer