PX4/PX4-Autopilot #12720: [drivers & uORB] Differentiate distance sensors by ID's (not only by the rotation)

TSC21, 2019-08-16:
@dagar @DanielePettenuzzo we continue to set the ID's to 0 every time we add a new distance_sensor. This represents a problem when we have two distance sensors, facing the same direction, connected to the same I2C bus: how exactly does one differentiate between them if not through the ID? And how can that be set manually in that case, if we want to say, for example, the sensor on the left side has the ID 1 and the sensor on the right side has ID 2?

I would say that not having the possibility of manually or automatically setting the sensor IDs on a system that is aiming for supporting multiple distance sensors pointing in different directions is more a bug than a feature, so I think we need to sort out a solution to this soon enough.

**Describe problem solved by the proposed feature**
Make possible to differentiate between different distance sensors, even if they are pointing the same direction.

**Describe your preferred solution**
I say that we could differentiate the sensors based on their address on the bus - but I guess that shouldn't be something that the user should be worried about. So could we identify the sensor ID's, plugged on the same bus, based on the order that they are plugged on that bus? Not sure how that works.

TSC21, 2019-08-16:
A good example of a driver where we could already bring this in is https://github.com/PX4/Firmware/pull/12575. Since these kind of sensors tend to be used in numbers and pointing different directions for multiple purposes, including collision prevention.

mhkabir, 2019-08-16:
@dagar and I wanted to get rid of this ID field, make uORB more lean and do 1 sensor per uORB multi-topic.

TSC21, 2019-08-16:
> @dagar and I wanted to get rid of this ID field, make uORB more lean and do 1 sensor per uORB multi-topic.

What's your suggestion to differentiate between sensor streams?

dagar, 2019-08-17:
One option is to treat them the way we've handled other sensors (accel, gyro, mag). Maintaining a table of unique device types (https://github.com/PX4/Firmware/blob/master/src/drivers/drv_sensor.h) and each instance then has a device id (device type + bus + address).

From that point the configuration of each distance sensor instance would be similar to the setting each external mag rotation.

DanielePettenuzzo, 2019-08-18:
@dagar we also need a mechanism to have multiple sensors of the same type on the same I2C bus. @TSC21 and I looked at the lidar lite and the startup sequence should be that all sensors are disabled (via GPIO to the lidar) and then we pull them up one by one so we can change their I2C address. Is there any driver that already does something similar?

TSC21, 2019-08-27:
@DanielePettenuzzo any progress on this matter?

TSC21, 2019-09-02:
First step in https://github.com/PX4/Firmware/pull/12864

TSC21, 2019-12-01:
This is still relevant and needs to be solved.

TSC21, 2020-03-24:
@dagar this is still to solve.